slic3r/Slic3r #4500: Max Volumetric Speed extruder-specific request

OrthogonalConcept, 2018-07-30:
1.3.1_dev (Repetier)

Windows 7 pro

### Behavior
*  Filament setting -max volumetric speed overide changes global setting

In my case, this global change prevents reasonable FDM part build time when using fine nozzle with TPU. Please make the max volumetric speed override extruder-specific instead of global.

I am using two different nozzle sizes in a dual extruder setup.  I need to print fine features with TPU from the smaller, 0.3mm nozzle onto a big chunk of PETG with a 0.5mm nozzle on the other extruder. The TPU has to be printed slowly, but the PETG can easily extrude 20 times faster with my setup. Since my max v speed override of 0.2mm^3/s for the TPU takes over the global setting, my part takes more than 100 hours to print. When I turn off the volumetric override, the part prints in 8 hours, but the feed is too fast for the TPU and I get crazy-lots-o-gaps in the fine features.

[Cell Wash Stopper.zip](https://github.com/slic3r/Slic3r/files/2242242/Cell.Wash.Stopper.zip)

Thanks.

lordofhyphens, 2018-07-30:
So the expected behavior should be that the maximum volumetric speed from a filament setting should only apply to extruders that use that filament profile?

OrthogonalConcept, 2018-07-30:
Yes. 
If there is an override entered for a filament profile, then it should apply to only the extruder using that profile.
If there are two different overrides, then each extruder should run with the limit specified in that profile
If there is no override for one extruder, then the extruder should default to the global limit.

Thanks.

lordofhyphens, 2018-07-30:
Ok, thank you. I wanted to be sure so I could write some tests against it :)

OrthogonalConcept, 2018-07-30:
My pleasure! Thanks for all you do.

Jardal, 2020-06-18:
Max volumetric speed is functional from Print settigns only, not from Filament.

From gcode:
```
; generated by Slic3r 1.3.1-dev (Build 54a31eed204863c49dfaa3b6fe4809f6d10a94ba) on 2020-06-18 at 13:59:58

; external perimeters extrusion width = 0.50mm (2.74mm^3/s)
; perimeters extrusion width = 0.50mm (3.50mm^3/s)
; infill extrusion width = 0.44mm (3.50mm^3/s)
; solid infill extrusion width = 0.50mm (3.50mm^3/s)
; top infill extrusion width = 0.50mm (1.83mm^3/s)

; external perimeters extrusion width = 0.50mm (2.74mm^3/s)
; perimeters extrusion width = 0.50mm (3.50mm^3/s)
; infill extrusion width = 0.44mm (3.50mm^3/s)
; solid infill extrusion width = 0.50mm (3.50mm^3/s)
; top infill extrusion width = 0.50mm (1.83mm^3/s)

; filament_max_volumetric_speed = 3,1.5
; max_volumetric_speed = 3.5
; use_volumetric_e = 0

```

RealET, 2020-07-05:
Any progress on this?  Or has anyone found a workaround?

lordofhyphens, 2020-07-06:
Can you specify the max volumetric speed in your firmware?

Jardal, 2020-07-06:
Workaround is none. Volumetric speed limit:

- is not working from Filament tab
- is working from Print tab

For dual color printing we need to differentiate between each volumetric speed for each specific filament

lordofhyphens, 2020-07-06:
Outside of Slic3r your firmware may provide a max speed in volume units per extruder.