slic3r/Slic3r #4502: [Feature Request]: display/change print order when "complete individual objects" is enabled.

AndKe, 2018-08-04:
When placing multiple objects, the order of printing is important, user needs to know the order, to avoid collision when travelling to next object.

Imagine three objects on X axis:
5cm_high , 10cm_high and 9cm_high. 

if 10cm_high is printed first, then 5cm, then 9cm_object, the extruder will crash into the 9cm item when travelling from 5cm to 10 cm.

By forcing, or knowing the order, one can optimize and place more objects on the same bed, safely.

By knowing this order, user could start to print objects from far way from docking point, and thus also be able to pause print/change filament for any object, as there would be nothing in the path to the parking area.

lordofhyphens, 2018-08-12:
[image]

Tooltip:

"Set this to alter object priority for sequential printing. Objects are first sorted by priority (smaller integers print first), then by height."

lordofhyphens, 2018-08-12:
@AndKe Please give it a try and if it works for you, could you draft a short update to https://github.com/slic3r/Slic3r-Manual? It's all in Markdown and I do accept pull requests.

closed 2018-08-12 by lordofhyphens

AndKe, 2018-08-13:
It does not help with objects there are more than one of:
example:
I add object a , then use "more" to make 8 copies of that, add another object, then make for example 8 of that. once all 16 objects are visible, I drag them around to minimize space between, having carriage size and asymmetric features of carriage in mind  - at this point, what's what can quickly become an unholy mess.
Your solution is good  for different objects, but multiple iterations of one, is not possible to sort out.

lordofhyphens, 2018-08-13:
"More" is treated differently than add/remove and so far as I am aware none of the modifier meshes, etc, can be applied to copies -- they're true copies in that they will always share the same parameters. 

Your immediate workaround is to "Add" more instead of hitting the "More" button.

AndKe, 2018-08-13:
yes.
Thanks for trying, I still hope this feature-request will stay as such, until somebody can solve it by displaying tooltip over them or other foolproof method.
- thank you for trying.

lordofhyphens, 2018-08-13:
@AndKe the use case described in your issue is the one that is already covered by the current code. Please open a new issue to cover that specific use case.