slic3r/Slic3r #4503: [BUG] Alert windows when using print modifier

macsddau, 2018-08-05:
### Version
1.2.9

### Operating system type + version
Mac OS 10.13.6

### Behavior
About 150 alert windows when using print modifier.

#### STL/Config (.ZIP) where problem occurs
[[BUG] Alert windows when using print modifier.zip](https://github.com/slic3r/Slic3r/files/2260554/BUG.Alert.windows.when.using.print.modifier.zip)

lordofhyphens, 2018-08-05:
Screenshot would be nice.

Also, grab a copy of 1.3.0.

https://github.com/slic3r/Slic3r/releases/tag/1.3.0

macsddau, 2018-08-05:
Very nice with 1.3.0, no more alert window.

Thanks a lot !

[image]

closed 2018-08-05 by lordofhyphens