slic3r/Slic3r #4508: Gcode to change temp not working

rchancey, 2018-08-10:
### Version
Slic3r 1.41.0 -beta+.

### Operating system type + version
Mac High Sierra 10.13.6 (17G65)

### Behavior
Joseph suggested I open a ticket here. My question really has two parts. I am trying to get ngen to print correctly (in fact in the forums there are a LOT of folks struggling with ngen). I got some ngen orange from Prusa .. I assume it is the color fabb. 

My bridges are super saggy and the tops are not smooth and the I posted a few times on the forums. 

Issues: 
1) I am trying to insert the code to change the extruder and bed temps (and eventually retraction length, bridge speed, etc). I am adding code to the 'Before layer change G-code' box. 

I added the code from this thing verse: https://www.thingiverse.com/thing:2729076

the code I tried was this: 
{if layer_z==1.6}
; T tower floor 1
M104 S250
{elsif layer_z==11.6}
; T tower floor 2
M104 S245
{elsif layer_z==21.6}
; T tower floor 3
M104 S240

Temp stayed at 240

_Is this a new feature request?_
The feature would be if there was an ini file that included some GCode for changing some of the 'standard' settings for a temp tower to test and tune a new filament.. then each person can just run the different towers and save a LOT of filament and time. I can make a few sets of code if I can get a cheatsheet for the parameters for changing the settings. Since the parameters are exposed in the Slic3r I assume they can be set via the command line. 

#### STL/Config (.ZIP) where problem occurs
See above

lordofhyphens, 2018-08-10:
Your conditional gcode is using the Prusa3D syntax, which is not the same as the regular Slic3r syntax.

https://github.com/slic3r/Slic3r-Manual/blob/master/src/advanced/conditional-gcode.md

I may accept a pull request to add a calibration script. 
Could you confirm the behavior on a build from https://dl.slic3r.org/dev/mac/ ?

rchancey, 2018-08-10:
Ok.. thanks Joseph. I will try that version.. I am currently printing again on the beta and it seems to have changed temp this time!. it is at 250 which is great! So it may have been a fluke somehow. 

Is there a Prusa3D cheat sheet on the different M codes for changing things like retraction length and speed? Same for bed temp.

rchancey, 2018-08-10:
Joseph, 
I found the retraction length code.. how do I know what codes work for the Prusa version of Slic3r?
and on the M207.. can I just put M207 S1 to change retraction from the default 0.9, to a 1mm?

A lot of the code seem to indicate 'deprecated' and to move to G10.. so not sure what to do about that other than try codes and see? 

[image]

lordofhyphens, 2018-08-10:
@rchancey Couldn't tell you offhand, I don't have a Prusa printer, nor do I use their fork of Slic3r. The "beta" you are referring to has its source code (and its own issue tracker) at http://github.com/prusa3d/Slic3r and we do not offer *any* support for it here. We're not Prusa3D here.

Our own "beta" builds are at https://dl.slic3r.org/dev/, https://github.com/slic3r/Slic3r/releases for releases. 

What you are asking about is whether your printer *firmware* supports those Gcode commands, not the slicer. Consult your firmware documentation (if it's a Prusa3D printer, then contact their support people as they have made significant changes to their printer firmware IIRC). 

 Closing issue for now; will reopen if you can reproduce problem with one of our builds and a valid config for us, not Prusa's.

closed 2018-08-10 by lordofhyphens