slic3r/Slic3r #4510: [FEATURE REQUEST] Ability to Export PDF of Print Configuration with gcode

PrestonL, 2018-08-14:
The ability to export a PDF of the print/filament/printer configuration would be amazing in being able to track print results when experimenting.  Even better would be an "Export G-code/PDF" where the PDF has a UID alongside the gcode file ( object-005.gco and object-005.pdf etc )

Even a basic way of doing this to track and tag prints would be amazing.  If PDF support would be too cumbersome to implement due to libraries/etc, even a basic HTML or XML/XSL fileset would suffice.  If HTML were used, an XML/XSL backend generating the HTML file would be great since adding the ability to export the XML/XSL could prove helpful in integrating to things like Octoprint and other third party utilities that might utilize and visualize the data.

Thoughts?

lordofhyphens, 2018-08-14:
The  Gcode already contains the entire print configuration in its postamble.

lordofhyphens, 2018-08-14:
Could you explain the use case differently, maybe with a picture? I'm having problems understanding what you are looking for. 

Is this the text of the configuration used to generate the print?

lordofhyphens, 2018-08-14:
This information is already present in the GCODE export. 

````
; filament used = 249.6mm (0.6cm3)
; total filament cost = 0.0

; avoid_crossing_perimeters = 0
; bed_shape = 79.5618x8.36228,78.2518x16.6329,76.0845x24.7214,73.0836x32.5389,69.282x40,64.7214x47.0228,59.4516x53.5304,53.5304x59.4516,47.0228x64.7214,40x69.282,32.5389x73.0836,24.7214x76.0845,16.6329x78.2518,8.36228x79.5618,0x80,-8.36228x79.5618,-16.6329x78.2518,-24.7214x76.0845,-32.5389x73.0836,-40x69.282,-47.0228x64.7214,-53.5304x59.4516,-59.4516x53.5304,-64.7214x47.0228,-69.282x40,-73.0836x32.5389,-76.0845x24.7214,-78.2518x16.6329,-79.5618x8.36228,-80x0,-79.5618x-8.36228,-78.2518x-16.6329,-76.0845x-24.7214,-73.0836x-32.5389,-69.282x-40,-64.7214x-47.0228,-59.4516x-53.5304,-53.5304x-59.4516,-47.0228x-64.7214,-40x-69.282,-32.5389x-73.0836,-24.7214x-76.0845,-16.6329x-78.2518,-8.36228x-79.5618,0x-80,8.36228x-79.5618,16.6329x-78.2518,24.7214x-76.0845,32.5389x-73.0836,40x-69.282,47.0228x-64.7214,53.5304x-59.4516,59.4516x-53.5304,64.7214x-47.0228,69.282x-40,73.0836x-32.5389,76.0845x-24.7214,78.2518x-16.6329,79.5618x-8.36228,80x0
; bed_temperature = 62
; before_layer_gcode = 
; between_objects_gcode = 
; bridge_acceleration = 5000
; bridge_fan_speed = 100
; brim_connections_width = 0
; brim_width = 0
; complete_objects = 0
; cooling = 1
; default_acceleration = 5000
; disable_fan_first_layers = 1
; duplicate_distance = 6
; end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
; end_gcode = M104 S0 ; turn off temperature\nM140 S0;\nG28 ;home all axis\nM84; disable motors\n
; extruder_clearance_height = 20
; extruder_clearance_radius = 20
; extruder_offset = 0x0
; extrusion_axis = E
; extrusion_multiplier = 1
; fan_always_on = 1
; fan_below_layer_time = 30
; filament_colour = #4EFF00
; filament_cost = 0
; filament_density = 0
; filament_diameter = 1.72
; filament_max_volumetric_speed = 0
; filament_notes = ""
; first_layer_acceleration = 5000
; first_layer_bed_temperature = 62
; first_layer_extrusion_width = 0
; first_layer_speed = 30
; first_layer_temperature = 236
; gcode_arcs = 0
; gcode_comments = 1
; gcode_flavor = smoothie
; has_heatbed = 1
; infill_acceleration = 5000
; infill_first = 0
; interior_brim_width = 0
; layer_gcode = 
; max_fan_speed = 100
; max_layer_height = 0.3
; max_print_speed = 80
; max_volumetric_speed = 6
; min_fan_speed = 100
; min_layer_height = 0.15
; min_print_speed = 20
; min_skirt_length = 5
; notes = 
; nozzle_diameter = 0.35
; only_retract_when_crossing_perimeters = 1
; ooze_prevention = 0
; output_filename_format = [input_filename_base].gcode
; perimeter_acceleration = 2500
; post_process = 
; pressure_advance = 0
; printer_notes = 
; resolution = 0
; retract_before_travel = 1
; retract_layer_change = 1
; retract_length = 2
; retract_length_toolchange = 10
; retract_lift = 0
; retract_lift_above = 0
; retract_lift_below = 0
; retract_restart_extra = 0
; retract_restart_extra_toolchange = 0
; retract_speed = 40
; skirt_distance = 2
; skirt_height = 1
; skirts = 1
; slowdown_below_layer_time = 60
; spiral_vase = 0
; standby_temperature_delta = -5
; start_filament_gcode = "; Filament gcode\n"
; start_gcode = M106 S0\nG90; set to absolute position\nG28\nM190 S60; preheat bed\nM104 S100; preheat nozzle.\nM190 S[first_layer_bed_temperature]; Wait until heatbed hits [first_layer_bed_temperature] \nG0 X0 Y0 Z0\nM109 S[first_layer_temperature]; heat nozzle to [first_layer_temperature]C;
; temperature = 237
; threads = 4
; toolchange_gcode = 
; travel_speed = 80
; use_firmware_retraction = 1
; use_relative_e_distances = 0
; use_set_and_wait_bed = 0
; use_set_and_wait_extruder = 0
; use_volumetric_e = 1
; vibration_limit = 0
; wipe = 0
; z_offset = 0
; z_steps_per_mm = 0
; adaptive_slicing = 0
; adaptive_slicing_quality = 75%
; dont_support_bridges = 0
; extrusion_width = 0
; first_layer_height = 0.3
; infill_only_where_needed = 1
; interface_shells = 0
; layer_height = 0.15
; match_horizontal_surfaces = 0
; raft_layers = 0
; regions_overlap = 0
; seam_position = aligned
; sequential_print_priority = 0
; support_material = 0
; support_material_angle = 0
; support_material_buildplate_only = 1
; support_material_contact_distance = 0.2
; support_material_enforce_layers = 0
; support_material_extruder = 1
; support_material_extrusion_width = 0
; support_material_interface_extruder = 1
; support_material_interface_extrusion_width = 0
; support_material_interface_layers = 3
; support_material_interface_spacing = 0.5
; support_material_interface_speed = 100%
; support_material_pattern = rectilinear-grid
; support_material_spacing = 1
; support_material_speed = 60
; support_material_threshold = 120%
; xy_size_compensation = 0
; bottom_infill_pattern = rectilinear
; bottom_solid_layers = 2
; bridge_flow_ratio = 1
; bridge_speed = 60
; external_perimeter_extrusion_width = 0
; external_perimeter_speed = 50%
; external_perimeters_first = 0
; extra_perimeters = 1
; fill_angle = 45
; fill_density = 20%
; fill_gaps = 1
; fill_pattern = cubic
; gap_fill_speed = 20
; infill_every_layers = 2
; infill_extruder = 1
; infill_extrusion_width = 0
; infill_overlap = 60%
; infill_speed = 80
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 60
; perimeters = 3
; small_perimeter_speed = 30
; solid_infill_below_area = 70
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0
; solid_infill_speed = 20
; thin_walls = 1
; top_infill_extrusion_width = 0
; top_infill_pattern = rectilinear
; top_solid_infill_speed = 25
; top_solid_layers = 3
````

lordofhyphens, 2018-08-14:
Alternatively, jam CTRL-E after you start your print and save the config and you'll have a complete backup of all the settings (capable of being reimported back into Slic3r).

7eggert, 2018-08-16:
I think @PrestonL would like a pretty-printed version of this and/or the ability to automatically save myprint-042.config.ini (or myprint-042.config.pdf) along with myprint-042.gcode, so you could prepare a number of tests, print them one by one (from stick) and not get confused. I can see general use for auto-saving the configs, since you can't load the gcode file as a config file.

Instead of pretty-printing the config, I'd suggest writing down the number on the prints (that's what I do), but that's said without complete insight into @PrestonL's use cases.

lordofhyphens, 2018-08-16:
Honestly, I'd rather support loading the cached config from gcode files (accepting that it can take a long time) than writing a pretty-printer for the configs.

foreachthing, 2018-08-30:
I just had some time ... time to try python for the first time.
Please try this one and let me know how you like it (or not): https://github.com/foreachthing/Slic3rConfigPrettyPrint

lordofhyphens, 2018-09-08:
@PrestonL give the script by @foreachthing a try.