slic3r/Slic3r #4513: Cannot start slic3r(-1.3.1-dev-4f3e89a-linux-x64.tar.bz2): "slic3r.pl": No such file or directory

7eggert, 2018-08-16:
Version slic3r-1.3.1-dev-4f3e89a-linux-x64 from https://dl.slic3r.org/dev/linux/

(tar -tvaf ... returns OK; contains one file "Slic3r", which I renamed to contain the version number)

When trying to start this, I get:
Can't open perl script "slic3r.pl": No such file or directory

section of strace:

readlink("/proc/self/exe", "/home/7eggert/.lib/slic3r-1.3.1-"..., 4095) = 53
getuid()                                = 1000
geteuid()                               = 1000
getgid()                                = 1000
getegid()                               = 1000
ioctl(0, SNDCTL_TMR_TIMEBASE or SNDRV_TIMER_IOCTL_NEXT_DEVICE or TCGETS, {B38400 opost isig icanon echo ...}) = 0
lseek(0, 0, SEEK_CUR)                   = -1 ESPIPE (Illegal seek)
ioctl(1, SNDCTL_TMR_TIMEBASE or SNDRV_TIMER_IOCTL_NEXT_DEVICE or TCGETS, {B38400 opost isig icanon echo ...}) = 0
lseek(1, 0, SEEK_CUR)                   = -1 ESPIPE (Illegal seek)
ioctl(2, SNDCTL_TMR_TIMEBASE or SNDRV_TIMER_IOCTL_NEXT_DEVICE or TCGETS, {B38400 opost isig icanon echo ...}) = 0
lseek(2, 0, SEEK_CUR)                   = -1 ESPIPE (Illegal seek)
open("slic3r.pl", O_RDONLY)             = -1 ENOENT (No such file or directory)
write(2, "Can't open perl script \"slic3r.p"..., 62Can't open perl script "slic3r.pl": No such file or directory

lordofhyphens, 2018-08-16:
I apparently broke the Linux auto builds, but I don't know how right now. I will look at it tonight.

lordofhyphens, 2018-08-20:
It's fixed now.

closed 2018-08-20 by lordofhyphens