slic3r/Slic3r #4516: Layer height > 1 becomes = 1, unless... set adaptive slice minimum - use adaptive slice.

MarlynAnderson, 2018-08-21:
### Version
1.3.1-dev
2018.08.20.1828.4572036.32bit

### Operating system type + version
Win10-64

### Behavior
It seems that entering a value of '2mm' ( or anything > 1) for a layer height (yes, big nozzle - 3.75mm) generates g-code with 1mm Z-steps.  Not a big issue as there is a work-around (and affects maybe .01% of users)... I got my home built clay-paste-printer a g-code file with 2mm layer height steps by setting the adaptive slice minimum to 2mm - and then used it.
I don't know if other parameters are affected - like volume following the '2' while only stepping up 1, because, well, it's clay paste.  I'm only beginning to 'dial it in' - don't know what perfect is yet.
Just sayin' - it shouldn't do that... step up Z by 1 when you entered a '2'.

#### STL/Config (.ZIP) where problem occurs
[Slic3r_clay-printer.zip](https://github.com/slic3r/Slic3r/files/2306465/Slic3r_clay-printer.zip)

lordofhyphens, 2018-08-21:
@MarlynAnderson does the preview step in 1mm or does it step in 2mm?

MarlynAnderson, 2018-08-26:
Found the root issue, but first - yes, Preview steps at 1mm - so volume put out is probably correct.
Here's what I noticed when I took a second to read the header info in the g-code file...
 Nozzle diameter was being set to 1.00mm.
So, am guessing that was limiting layer height to 1.00mm (not allowed to be greater)?
Because I wanted the extrusion width to = nozzle diameter (3.75) , I had been using 'Advanced' print settings and entering '100%' for everything.
Was I wrong to think that Slic3r would know that 100% of 3.75mm IS 3.75mm?
Yes.
I thought it 'perhaps too coincidental' that the nozzle diameter of 1.00mm had the same numbers in it as '100%' - and tried 101%.
Guess what! Nozzle diameter was set to 1.01mm!
So, it seems the text box entry value parser of 'mm' vs. '%' is not doing its job properly.
Not sure how the '101' becomes '1.01', but it does.
Work-around #2...  enter '3.75mm' for all extrusion widths.
Then, 2mm is accepted as a valid layer height and all is well.

lordofhyphens, 2018-08-27:
In the future when posting configurations, post a ctrl-E config, not a config bundle.

MarlynAnderson, 2018-08-27:
I was wondering about that  (TMI?)  - good to know.
I have an 'Agh!' to report. Work-around #2 is false.
I was playing with Slic3r at home with 'Adaptive Slicing' un-checked.
Then, I go to my 'lab with internet' and try the manual 'mm' width method - but on that computer I forgot to un-check 'Use Adaptive'. 
Without Adaptive slicing, and extrusion widths = 3.75mm,  it reverted back to 1mm layer height - so, no work-around #2.
I'm including a truncated g-code file showing...
 nozzle_diameter = 3.75, ext. widths all 3.75mm, adaptive_slicing=0, layer_height=2, first_layer_height= 2 (which it does!) - but all subsequent z-steps are 1mm.

[3_75_adap_off.zip](https://github.com/slic3r/Slic3r/files/2324563/3_75_adap_off.zip)

printerbuilder, 2018-09-12:
found the same issue and used the same workaround. wanted to print 2mm layer with a 6mm nozzle.
had to use adaptive slicer with a minimum and maximum of 2mm
Nozzle size setting in advanced area. I left general nozzle size in 'printer settings' at 3mm ( that was never an issue before, is that wrong?)

haslingen, 2019-03-29:
Hi,
I got 10 mm layer hight... But I want 20 ... What did I do wrong?

Settings

PRint setting
Layer hight 20 mm
I put in first layer 20 mm
Use adaptive 

Advanced first layer 100%
Defalt extrusion with = 30 mm

Filament settings
Diameter 30 mm

Printer setting
Extruder 1
Nozzle diameter 30
Min 20 mm
Max 20 mm

----

; generated by Slic3r 1.3.0 on 2019-03-29 at 22:47:00

; external perimeters extrusion width = 30.00mm (893.56mm^3/s)
; perimeters extrusion width = 30.00mm (1787.12mm^3/s)
; infill extrusion width = 30.00mm (2382.83mm^3/s)
; solid infill extrusion width = 30.00mm (595.71mm^3/s)
; top infill extrusion width = 30.00mm (446.78mm^3/s)

G28 ; home all axes
G1 Z5 F1800 ; lift nozzle

; Filament gcode

G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 Z20.000 F3600.000
G1 E-2.00000 F2400.00000
G92 E0
G1 X1110.000 Y1350.000 F3600.000
G1 E2.00000 F2400.00000
G1 F3600
G1 X1060.000 Y1350.000 E24.22222
G1 Z30.000 F3600.000
G1 X1110.000 Y1350.000 F3600.000
G1 F1800
G1 X1060.000 Y1350.000 E36.85133
G1 Z40.000 F3600.000
G1 X1110.000 Y1350.000 F3600.000
G1 F1800
G1 X1060.000 Y1350.000 E49.48044
G1 Z50.000 F3600.000
G1 X1110.000 Y1350.000 F3600.000
G1 F1800
G1 X1060.000 Y1350.000 E62.10955
G1 Z60.000 F3600.000
G1 E60.10955 F2400.00000
G92 E0
G1 X1110.000 Y1350.000 F3600.000
G1 E2.00000 F2400.00000
G1 F1800
G1 X1060.000 Y1350.000 E14.62911
G1 E12.62911 F2400.00000
G92 E0
; Filament-specific end gcode 
;END gcode for filament

G28 X0  ; home X axis
M84     ; disable motors

M104 S0 ; set temperature
; filament used = 72.7mm (51.4cm3)
; total filament cost = 0.0

; avoid_crossing_perimeters = 0
; bed_shape = 0x0,2170x0,2170x2700,0x2700
; bed_temperature = 0
; before_layer_gcode = 
; between_objects_gcode = 
; bridge_acceleration = 0
; bridge_fan_speed = 100
; brim_connections_width = 0
; brim_width = 0
; complete_objects = 0
; cooling = 0
; default_acceleration = 0
; disable_fan_first_layers = 3
; duplicate_distance = 6
; end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
; end_gcode = G28 X0  ; home X axis\nM84     ; disable motors\n
; extruder_clearance_height = 20
; extruder_clearance_radius = 20
; extruder_offset = 0x0
; extrusion_axis = E
; extrusion_multiplier = 1
; fan_always_on = 0
; fan_below_layer_time = 0
; filament_colour = #B7BEC8
; filament_cost = 0
; filament_density = 0
; filament_diameter = 30
; filament_max_volumetric_speed = 0
; filament_notes = ""
; first_layer_acceleration = 0
; first_layer_bed_temperature = 0
; first_layer_extrusion_width = 100%
; first_layer_speed = 60
; first_layer_temperature = 0
; gcode_arcs = 0
; gcode_comments = 0
; gcode_flavor = smoothie
; has_heatbed = 0
; infill_acceleration = 0
; infill_first = 0
; interior_brim_width = 0
; layer_gcode = 
; max_fan_speed = 100
; max_layer_height = 20
; max_print_speed = 80
; max_volumetric_speed = 0
; min_fan_speed = 35
; min_layer_height = 10
; min_print_speed = 10
; min_skirt_length = 0
; notes = 
; nozzle_diameter = 30
; only_retract_when_crossing_perimeters = 1
; ooze_prevention = 0
; output_filename_format = [input_filename_base].gcode
; perimeter_acceleration = 0
; post_process = 
; pressure_advance = 0
; printer_notes = 
; resolution = 0
; retract_before_travel = 2
; retract_layer_change = 0
; retract_length = 2
; retract_length_toolchange = 10
; retract_lift = 0
; retract_lift_above = 0
; retract_lift_below = 0
; retract_restart_extra = 0
; retract_restart_extra_toolchange = 0
; retract_speed = 40
; skirt_distance = 6
; skirt_height = 1
; skirts = 0
; slowdown_below_layer_time = 5
; spiral_vase = 0
; standby_temperature_delta = -5
; start_filament_gcode = "; Filament gcode\n"
; start_gcode = G28 ; home all axes\nG1 Z5 F1800 ; lift nozzle\n
; temperature = 0
; threads = 4
; toolchange_gcode = 
; travel_speed = 60
; use_firmware_retraction = 0
; use_relative_e_distances = 0
; use_set_and_wait_bed = 0
; use_set_and_wait_extruder = 0
; use_volumetric_e = 0
; vibration_limit = 0
; wipe = 0
; z_offset = 0
; z_steps_per_mm = 0
; adaptive_slicing = 1
; adaptive_slicing_quality = 75%
; dont_support_bridges = 1
; extrusion_width = 30
; first_layer_height = 20
; infill_only_where_needed = 0
; interface_shells = 0
; layer_height = 20
; match_horizontal_surfaces = 0
; raft_layers = 0
; regions_overlap = 0
; seam_position = aligned
; sequential_print_priority = 0
; support_material = 0
; support_material_angle = 0
; support_material_buildplate_only = 0
; support_material_contact_distance = 0.2
; support_material_enforce_layers = 0
; support_material_extruder = 1
; support_material_extrusion_width = 0
; support_material_interface_extruder = 1
; support_material_interface_extrusion_width = 0
; support_material_interface_layers = 3
; support_material_interface_spacing = 0
; support_material_interface_speed = 100%
; support_material_max_layers = 0
; support_material_pattern = pillars
; support_material_spacing = 25
; support_material_speed = 60
; support_material_threshold = 60%
; xy_size_compensation = 0
; bottom_infill_pattern = rectilinear
; bottom_solid_layers = 1
; bridge_flow_ratio = 1
; bridge_speed = 60
; external_perimeter_extrusion_width = 0
; external_perimeter_speed = 50%
; external_perimeters_first = 0
; extra_perimeters = 1
; fill_angle = 45
; fill_density = 20%
; fill_gaps = 1
; fill_pattern = stars
; gap_fill_speed = 20
; infill_every_layers = 1
; infill_extruder = 1
; infill_extrusion_width = 0
; infill_overlap = 55%
; infill_speed = 80
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 60
; perimeters = 1
; small_perimeter_speed = 15
; solid_infill_below_area = 70
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0
; solid_infill_speed = 20
; thin_walls = 1
; top_infill_extrusion_width = 0
; top_infill_pattern = rectilinear
; top_solid_infill_speed = 15
; top_solid_layers = 1

lordofhyphens, 2019-04-11:
@MarlynAnderson sorry it took this long for me to get to it, but I have a PR available to fix this issue.
The routine used to get the list of Z layers to slice at (not using adaptive slicing) assumed that everyone would use nozzles with diameter < 1.0mm so that value got used as a ceiling (and 10mm layer heights were considered a pipe dream @haslingen).

I've pushed both of those ceilings to the limit. Literally: `std::numeric_limits<double>::max()` is now the starting point for the algorithm, which the numerical maximum for the platform it's compiled on. If you want/need something bigger than that, it can't happen.

lordofhyphens, 2019-04-11:
@haslingen there's a 120mm cube sliced with your config.
[image]

closed 2019-04-13 by lordofhyphens

haslingen, 2019-11-14:
Hi,

I have an other question. I made my print bed 5000 x 2170 x 1800 mm. Then I made a shape that is 4500 x 2000 x 500 mm (length, with, hight). I got a problem. The slicer makes the length very small. I would gess maby changes 5000 to 500 mm Is there an error in the printbed size?

/T

> @MarlynAnderson <https://github.com/MarlynAnderson> sorry it took this long for me to get to it, but I have a PR available to fix this issue.
> The routine used to get the list of Z layers to slice at [...]

Vänliga hälsningar
Tobias v. Haslingen

Tobias v. Haslingen
Haslingen Consulting
Bergavägen 10 
182 53 Danderyd
+46 702 45 76 77
tobias@haslingen.com

PR #4780 (2019-04-13): Set the initial maximum size for layer heights and nozzle diameter to max
The algorithm used to adjust layer heights in PrintObject has as a ceiling some default values that were fine when nozzles were < 1.0mm but not anymore.
To avoid more surprises, set to numeric_limits<double>::max()
Adds regression test as well.
Fixes #4516
commit 371d87ce (2019-04-13): Set the initial maximum size for layer heights and nozzle diameter to max
The algorithm used to adjust layer heights in PrintObject has as a ceiling some default values that were fine when nozzles were < 1.0mm but not anymore.
To avoid more surprises, set to numeric_limits<double>::max()
Adds regression test as well.
Fixes #4516