slic3r/Slic3r #4520: Exterior brim causes incorect printing of support materials

Lebensgefahr, 2018-08-24:
### Version
1.3.0-dev 
904baff
### Operating system type + version
Windows 7 64

### Behavior
If i turn on exterior brim the first layer of support material doesn't print at all and this causes that the next layer of support material have no basis to melt with.
Exterior brim width = 0
[image]

Exterior brim width = 5
[image]

MortarArt, 2018-08-28:
This may be related to; https://github.com/slic3r/Slic3r/issues/4427

mrvanes, 2018-11-28:
I also see this problem. 
Build commit: 26b2a1e0cd36753c9f6e05609a5658cb91ab6727

These pictures may clarify what happens with and without external brim.

3mm exterior brim, layer 1. Pay attention to support in the upper left circle, (exactly) the brim width is missing:
[image]

3mm exterior brim, layer 2. Next suppor layer is deposited "in air":
[image]

0mm exterior brim, layer 1:
[image]

0mm exterior brim, layer 2:
[image]

superzerg, 2021-01-16:
I have this problem as well (Slic3r v 1.3):
[image]

Lebensgefahr, 2021-01-17:
I found some interesting moment. If your layer high is 0.2 mm but your first layer high is 50% (0.1) it makes detail higher for 0.1 mm.
I will try to explain. If your detail is 0.2 mm high and first layer is 50% of layer high than you will get detail with 0.3 mm high. But if your first layer high is 100% than you will get detail with 0.2 mm high. You can check this by running an easy test.
And i think it will cause problems in support material printing for the first layer. 
Slic3r has a lot of bugs and it is not supported as far as i can see. I reported another bug few weeks ago.