slic3r/Slic3r #4522: Raft printed with incorrect extruder.

MortarArt, 2018-08-28:
### Version
_Version of Slic3r used goes here_
1.3.1-dev

### Operating system type + version

Win10-64bit

### Behavior
Raft printed with incorrect extruder.

[image]

Multi-extruder settings.

[image]

Result preview in Repetier.

lordofhyphens, 2018-08-28:
Please upload the Ctrl-E config export and a test model.

MortarArt, 2018-08-28:
[config_and_model.zip](https://github.com/slic3r/Slic3r/files/2328218/config_and_model.zip)

lordofhyphens, 2018-09-01:
Thank you!

Also, as a FYI, both the 3MF and AMF exports will save any overrides set via per-object settings (right-click in the plater).

lordofhyphens, 2018-09-01:
After some brief investigation (thank you for the config export!), I was able to reproduce partially. 
The config you sent me had no raft layers (raft != brim), so I added them and got this result:
[image]
I colored the toolpaths by filament chosen.

config export: 
[config_raft.zip](https://github.com/slic3r/Slic3r/files/2342054/config_raft.zip)

Turning on verbose gcode confirmed that it was indeed changing to extruder 2 prior to printing the brim.

[image]

However, with 0 raft layers there's just a brim and it prints with the perimeter extruder instead of the support/raft extruder. AFAIK this is generally expected but I can see situations where it is not ideal. 

I have edited the title of the issue accordingly.

lordofhyphens, 2018-09-01:
````
    // brim is only printed on first layer and uses perimeter extruder
````
It's intentional ;)

I guess I'm taking suggestions for what to do about this.

lordofhyphens, 2018-09-01:
The flow is specifically set in Print::brim_flow() and defaults to the perimeter extrusion there.

I don't actually know why it's getting printed with the support extruder when there is a raft. I suspect it's just not getting reset.