slic3r/Slic3r #4523: Crashes in Ubuntu 18.04

GettingNowhere, 2018-08-30:
### Version
slic3r (1.2.9+dfsg-9)

### Operating system type + version
Ubuntu 18.04 LTS 
Kernel Linux 4.15.0-33-generic-x86_64
Mate 1.20.1

### Behavior
Worked in 16.04
GUI comes up for <.5 seconds
then aborts.
```
alistair@Main:~$ slic3r
(slic3r:18729): Gtk-WARNING **: 17:05:59.674: Theme parsing error: gtk-main.css:73:38: Failed to import: Error opening file /usr/share/themes/Radiance/gtk-3.20/apps/gnome-builder.css: No such file or directory
(slic3r:18729): Gtk-WARNING **: 17:06:00.974: Negative content height -11 (allocation 1, extents 6x6) while allocating gadget (node button, owner GtkToggleButton)
(slic3r:18729): Gtk-WARNING **: 17:06:00.974: Negative content width -13 (allocation 1, extents 7x7) while allocating gadget (node button, owner GtkButton)
(slic3r:18729): Gtk-WARNING **: 17:06:00.974: Negative content height -11 (allocation 1, extents 6x6) while allocating gadget (node button, owner GtkButton)
(slic3r:18729): Gtk-WARNING **: 17:06:00.974: Negative content width -13 (allocation 1, extents 7x7) while allocating gadget (node button, owner GtkButton)
(slic3r:18729): Gtk-WARNING **: 17:06:00.974: Negative content height -11 (allocation 1, extents 6x6) while allocating gadget (node button, owner GtkButton)
(slic3r:18729): Gtk-CRITICAL **: 17:06:00.974: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
(slic3r:18729): Gtk-CRITICAL **: 17:06:00.976: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
(slic3r:18729): Gtk-WARNING **: 17:06:00.977: Negative content height -11 (allocation 1, extents 6x6) while allocating gadget (node button, owner GtkToggleButton)
(slic3r:18729): Gtk-WARNING **: 17:06:00.978: Negative content width -13 (allocation 1, extents 7x7) while allocating gadget (node button, owner GtkButton)
(slic3r:18729): Gtk-WARNING **: 17:06:00.978: Negative content height -11 (allocation 1, extents 6x6) while allocating gadget (node button, owner GtkButton)
(slic3r:18729): Gtk-WARNING **: 17:06:00.978: Negative content width -13 (allocation 1, extents 7x7) while allocating gadget (node button, owner GtkButton)
(slic3r:18729): Gtk-WARNING **: 17:06:00.978: Negative content height -11 (allocation 1, extents 6x6) while allocating gadget (node button, owner GtkButton)
(slic3r:18729): Gtk-CRITICAL **: 17:06:00.978: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
(slic3r:18729): Gtk-CRITICAL **: 17:06:00.979: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
(slic3r:18729): Gtk-WARNING **: 17:06:00.980: Negative content height -11 (allocation 1, extents 6x6) while allocating gadget (node button, owner GtkToggleButton)
(slic3r:18729): Gtk-WARNING **: 17:06:00.980: Negative content width -13 (allocation 1, extents 7x7) while allocating gadget (node button, owner GtkButton)
(slic3r:18729): Gtk-WARNING **: 17:06:00.980: Negative content height -11 (allocation 1, extents 6x6) while allocating gadget (node button, owner GtkButton)
(slic3r:18729): Gtk-WARNING **: 17:06:00.980: Negative content width -13 (allocation 1, extents 7x7) while allocating gadget (node button, owner GtkButton)
(slic3r:18729): Gtk-WARNING **: 17:06:00.980: Negative content height -11 (allocation 1, extents 6x6) while allocating gadget (node button, owner GtkButton)
(slic3r:18729): Gtk-CRITICAL **: 17:06:00.980: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
(slic3r:18729): Gtk-CRITICAL **: 17:06:00.981: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
(slic3r:18729): Gtk-CRITICAL **: 17:06:00.981: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
(slic3r:18729): Gtk-CRITICAL **: 17:06:00.982: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
Segmentation fault (core dumped)
alistair@Main:~$ 
```
#### STL/Config (.ZIP) where problem occurs
No files loaded

lordofhyphens, 2018-09-01:
This is probably an Ubuntu problem, not a Slic3r problem.
Use the 1.3.0 AppImage and let me know if it is similarly affected, please.

lordofhyphens, 2018-09-01:
I know that there is a 1.3.0 package available for Debian sid. @hyperair any other thoughts?

lordofhyphens, 2018-09-08:
Closing as the current appimage works, the package maintainer has been tagged, and it's for an old, packaged version.

closed 2018-09-08 by lordofhyphens