slic3r/Slic3r #4526: Cannot see the Controller Tab

wheemzy1, 2018-09-01:
### Version
 1.3.1

### Operating system type + version

Ubuntu 16.04 & 18.04

### Behavior

I have a Rambo board. I was able to test the connection to Slic3r which succeeded. However, when I try to see the Control Window so that I can use the terminal to enter some GCode commands I cannot see any tab at all. No window will appear. I ran the program with and without sudo. 

Receive these errors, not sure if related. 

(slic3r.pl:4425): IBUS-WARNING **: The owner of /home/matthew/.config/ibus/bus is not root!

** (slic3r.pl:4425): CRITICAL **: murrine_style_draw_box: assertion 'width >= -1' failed

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

wheemzy1, 2018-09-01:
I am an idiot, it's in the settings, not sure how I missed it.

closed 2018-09-01 by wheemzy1

lordofhyphens, 2018-09-02:
These things happen ;)