slic3r/Slic3r #4527: error g-code generation

AndrewT1000, 2018-09-02:
### Version
1.3.1 dev - several versions from github 

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Ubuntu, Win 7

### Behavior
On end of print head/nozzle it's going down instead of going up a little. The printout is pressed against the bed and only then the head goes to the 0.0 position I discovered it for now on two models. The situation is repeatable. In this case model is on 180 % size
[Archive.zip](https://github.com/slic3r/Slic3r/files/2343164/Archive.zip)

foreachthing, 2018-09-02:
Maybe enter a real value in your custom footer gcode and use relative values (`G91`):
Change
`G1 Z+50 E-5 F{travel_speed}`
to 
```
G91
G1 Z+50 E-5 F6000
```

These are some of my last few lines:
```
G91 ; relative coordinates
G1 Z+15  X-10 Y-10 E-7  F6000
```

lordofhyphens, 2018-09-02:
I've also added it to the C++ CLI route (and fixed an unrelated bug where end gcode wasn't output).