slic3r/Slic3r #4531: slic3r infills a hole

l29ah, 2018-09-05:
### Version
880e9a52c34943ba4ee22a3d6c2162d314af4e66

### Operating system type + version
Gentoo GNU/Linux

### Behavior
[image]
[image]

#### STL/Config (.ZIP) where problem occurs
[bug.zip](https://github.com/slic3r/Slic3r/files/2354532/bug.zip)

foreachthing, 2018-09-06:
Use https://service.netfabb.com/service.php with your stl and you're all set.
[image]
Left: the fixed stl. Right: your original.

l29ah, 2018-09-06:
Interesting. Calling `admesh` on it doesn't fix it.

lordofhyphens, 2018-09-06:
If admesh could fix it, slic3r wouldn't show any issues.

closed 2018-09-06 by lordofhyphens

l29ah, 2018-09-07:
But i don't have any horizontal surfaces there.

l29ah, 2018-09-07:
Is there a solution that doesn't involve using magic web services?

lordofhyphens, 2018-09-07:
You can probably spot the issue with either mesh lab or the application version of netfabb.

l29ah, 2018-09-07:
On Fri, Sep 07, 2018 at 12:08:47PM -0700, Joseph Lenox wrote:
> You can probably spot the issue with either mesh lab or the application version of netfabb.

meshlab opens it just fine.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

l29ah, 2018-09-07:
[image] does this look bad?

By the way, the model comes from openscad, and i wasn't doing anything nasty with polygons, just subtracting primitives and hulls. Even the preview looks fine.

> manually

uhh

foreachthing, 2018-09-07:
Yes it does look bad!
Non manifold edges .. 54; I've seen worse!
Zero faces: 2209! WOW!
Intersecting faces: 484 .... baah!

Manually or use some magic service ;-)
In Blender it has a "make manifold" button. Also, if you enter the edit mode, while the part is selected (press TAB after "check all"), then you get more tools to fix the mesh.

foreachthing, 2018-09-08:
Maybe tweak the settings in openscad ... reduce or increase tessellation until you get a good model.