slic3r/Slic3r #4536: Windows build with CMake and VS 2017

nszeuss, 2018-09-09:
### Operating system type + version
Windows 7 64

### Behavior
I am using the build with cmake and VS as described in:
 [https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#building-with-cmake-and-ms-visual-studio-2017-community-edition](url)
I use VS 2017 community 15.8.3.

build succeeded, however the slic3r.exe crashes when slicing every STL I upload.
It crashes on some std::vector methods.
I compare this behaviour to a build with Perl, and it seems like Perl C libs has some overriding for std::vector, and therefore the different behaviour.

Does build with cmake and VS has any other dependencies other than boost 1.67.0? how can I build it and make it run with VS?

lordofhyphens, 2018-09-10:
I've done a build with VS 2017, but I haven't tested it much at all (I rarely do any development for Slic3r on windows, and I've only tested using VS2017 as far as compilation).

It would be helpful to trace through the stack and figure out where it is failing; the code paths for the Perl and C++ are very different when it comes to slicing. You could be tripping over bugs in the PrintObject implementation.

nszeuss, 2018-09-13:
Thanks for the input.

I fixed the first exception I had regarding std::upper_bound in TriangleMeshSlicer<A>::_slice_do().
As I saw on call stack, std::upper_bound implementation is different with and without Strawberry Perl.

I fixed it by changing the code to :
	max_layer = std::upper_bound(z.begin() + (min_layer - z.begin()), z.end(), max_z);

	int max_idx = max_layer - z.begin();
	if (max_idx > 0)
		max_layer -= 1;

instead of: max_layer = std::upper_bound(z.begin() + (min_layer - z.begin()), z.end(), max_z) - 1; // last layer whose slice_z is <= max_z

(With Strawberry it takes the implementation from \Strawberry\c\lib\gcc\x86_64-w64-mingw32\6.3.0\include\c++\bits\stl_alfo.h)

Now I get another exception of access violation reading in PolyTree::Clear() (attached image).
I couldn't figure out why. Attached is the call stack. any idea how I can debug it?

[image]
[call_stack.txt](https://github.com/slic3r/Slic3r/files/2379311/call_stack.txt)