slic3r/Slic3r #4537: Question: Running Python Script in `Post-Processing Scrips`

foreachthing, 2018-09-10:
### Version
1.3.1 dev

### Operating system type + version
Win 7/8.1 x64

### Behavior
* Adding full path to python file with parameters in `Post-Processing Script`
  * `c:\temp\Slic3rConfigPrettyPrint.py -au "foreaththing"`
* Export GCode
* **Notepad++ opens said python file**

* Expected Result:
  * Slic3r runs python file with parameters

### Question:
**How or in what form do I put the file path in `Post-Processing Script` that Slic3r _runs_ the python script?**

#### STL/Config (.ZIP) where problem occurs
* Python script: https://github.com/foreachthing/Slic3rConfigPrettyPrint/blob/master/Slic3rConfigPrettyPrint.py
* STL: use your favorite model
* Config: add path to the script to `Post-Processing Script` in `Print Settings`

lordofhyphens, 2018-09-10:
If the default action for "executing" a Python script isn't to run python.exe on your windows system, you'd need to provide the full path to Python and then the script as a single line.

This is an inherent problem with Windows.

closed 2018-09-10 by lordofhyphens

lordofhyphens, 2018-09-10:
@foreachthing would you be willing to submit a PR to the Manual with this information and/or other related experiences?