slic3r/Slic3r #4542: The controller tab is missing from the interface

7eggert, 2018-09-18:
### Version 3.1.0-dev d17659b67c65bb5003ec92bf0d56ed7f8810e970
### Debian Jessie / Linux
### Behavior
The Controller tab is missing from the user interface. Instead any tab being at the second position (e.g. printer settings) will be activated on ^Y

lordofhyphens, 2018-09-18:
Controller tab cam be turned on/off from preferences, is this set to be shown (and reset)?

Assuming that you are referring to the normal Perl interface

7eggert, 2018-09-19:
a) Thanks. This seems to be a new feature; using an existing config from 1.3.0 seems to cause the setting to be unset.

b) You are referring to the Perl interface vs. the command line interface? Then obviously.

Changed topics of this issue:

- **Config import causes controller tab to disappear (wrong default while importing)**
- ^Y activates wrong tab then; the menu item should be grayed out and have a tooltip
- (Afterthought, not part of this bug report: ^Y might cause the controller tab to be displayed, too.)

lordofhyphens, 2018-09-19:
Reference: #4526 

Feature was introduced at #3984

Additionally, it shouldn't be set or unset by a print configuration, but by Slic3r's own GUI ini file (located in `~/.Slic3r` on posix systems)

lordofhyphens, 2018-09-19:
@7eggert there's a C++ GUI that is mostly nonfunctional (I haven't piped the configuration system through it yet); I was making sure you weren't messing around with that.

7eggert, 2018-09-19:
To clarify: I was talking about the GUI's ini file being left from 1.3.0. (But all config files remain untouched by upgrading, off cause).

alranel, 2018-11-05:
As of now:

* controller is disabled by default
* the *Window > Controller* menu item is always enabled
* selecting that menu item does nothing

I thought it is a bug. Then I found this issue and realized there's a Preference option to enable it again.

I propose the following behavior:

* show the Controller by default (so that people realize Slic3r has that feature)
* when the Controller is disabled and user selects *Window > Controller*, prompt user for reactivation of the preference, do it and tell user to restart Slic3r

lordofhyphens, 2018-11-05:
Should be fixed via #4575

closed 2018-11-05 by lordofhyphens