slic3r/Slic3r #4543: Custom gcode does not allow "\"

7eggert, 2018-09-18:
### Version 3.1.0-dev d17659b67c65bb5003ec92bf0d56ed7f8810e970
### Debian Jessie / Linux
### Behavior
If you write a backslash into the custom gcode field e.g. of the Start G-code field on the printer settings, then export the gcode file, the backslash will be stripped from the generated file.

I have no clue why anybody should use backslashes there, but that's what I found when I tested the pretty-print script of @foreachthing (created for #4510).

lordofhyphens, 2018-09-18:
I think this came out of avoiding a crash.

lordofhyphens, 2018-10-14:
Yeah I was avoiding a crash as in #4021 with that behavior.

lordofhyphens, 2018-10-27:
@7eggert 84275d08d218aa71bb6fc511e0acc8b1d6efbd59 is what introduced this behavior. I'll see what I can do.