slic3r/Slic3r #4544: Next X Y Z placeholders?

KickerTwin, 2018-09-21:
Are there any placeholders for getting the next x y or z coordinates for use in custom the gcode box? Something like {if **[next_z]**==10} G1 bla bla bla

Sorry if my format for asking is either in the wrong place or just straight up wrong all round, but I couldn't find anything that relates to this issue.

lordofhyphens, 2018-09-21:
I can say it would double the post-process time; look ahead like that is not generally feasible without taking multiple passes through the gcode.

KickerTwin, 2018-09-21:
Ok cool. Thank you very much! 

Just one more thing; how is nozzle lift managed? Is it just worked into the Z coordinates based on the "nozzle lift before travel" (or something like that) setup value?

lordofhyphens, 2018-09-21:
Nozzle lift is implemented in [GCodeWriter::lift()](https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/GCodeWriter.cpp#L539); the system maintains an internal state of how much it has been lifted.

KickerTwin, 2018-09-22:
Thanks for the explanation Joseph.
Would it be at all possible to somehow copy the next Z position just as a value without any logic included? With "next" in this case being defined as the Z position immediately following a tool change marker (T0, T1 etc) 

Im looking to develop a custom prime tower within the custom gcode function and the "next Z position" is the one the print head needs to do this properly within Slic3r. 

v -- Blasphemy to follow -- v  
I can do this "next Z" thing in Simplify3D, but would prefer to use Slic3r for many of the prints I do.

KickerTwin, 2018-09-23:
So I basically JUST need a reference to the Z height to work.
I read about the [layer_num] and [layer_z] placeholders and I think I can employ these to do the job, but I cant get them to work.
 In the custom gcode box Ive treid "layer_z", "[layer_z]", "{layer_z}" and even "{[layer_z]}". All of these just end up being printed as a text in the gcode output. I can do a logical operation like {if layer_z = 0.2}xxxxx, but even that doesn't work. The "xxxxx" is still output regardless of the validity of the evaluation.

Im sure Im doing something wrong but I cant figure out what it is!!

lordofhyphens, 2018-09-23:
http://manual.slic3r.org/advanced/conditional-gcode