slic3r/Slic3r #4545: Gcode parser

KickerTwin, 2018-09-22:
This is probably impossible, but Ill ask it anyway:

It is at all possible to employ Slic3r to look at a piece of gcode and perform modification operations as defined by the {if [current_extruder]} or {if [previous_extruder]} conditional placeholders? 

Some background: In Slic3r PE, when using their prime tower generator thing, the custom gcode box for "tool change" is ignored. So for example, I would like to use PE to make a gcode file with prime tower and then send it through Slic3r to insert custom gcode based on tool change commands.

Totally crazy?

lordofhyphens, 2018-09-22:
Probably better done as a post process script.

lordofhyphens, 2018-09-22:
https://github.com/slic3r/Slic3r/wiki/Writing-post-processing-scripts
http://manual.slic3r.org/advanced/post-processing

lordofhyphens, 2018-09-22:
The general interface for a post-processing "script" (this really can be anything) is a program that reads text from standard input and puts the transformed gcode on standard output. You have some environment variables available for use as well.