slic3r/Slic3r #4546: Overhang threshold setting NOT saved & CTRL+1 Don't open Print Setting

Lemmone, 2018-09-24:
1.3.0-dev (2018-09-08)
d17659b67c65bb5003ec92bf0d56ed7f8810e970

S.O. : Linux Mint 18.2 Cinnamon 64 bit

I have noticed that Overhang threshold is NOT saved and CTRL+1 don't open Print setting

Steps to reproduce:
1) Open any STL
2) Click on settings/Print settings (CTRL+1 don't open the page)
3) Click on Support material
4) Click on Generate support material
5) Change Overhang threshold to 0 for example
6) Click on blue floppy disk and give a name
7) Exit software
8) Reopen software
9) Follow from 1 to 4
10) Select saved setting in combo-box
11) Overhang threshold is at default value

Thank you, bye
Stefano

Lemmone, 2018-10-04:
Uhm....   No answer in 10 days :-((

lordofhyphens, 2018-10-04:
Yeah, extremely busy in my day job.