slic3r/Slic3r #4547: [request]

Liger0, 2018-09-26:
### Version
1.3.1 -dev
### Operating system type + version
Windows 10 x64

### Feature request
Since using slic3r I have a really horrible top layer compared to what I have in Cura using the top coat function which adds an additional layer on the standard top layers making the surface completely flat instead of being extremely rough like with slic3r, at least having the possibility to specify the top and bottom layers infill angle would be really good, thus I can decide which one could be more appealing with the rest of the walls.

lordofhyphens, 2018-09-27:
Are you talking about the "ironing" method?

supermerill, 2018-09-27:
> top and bottom layers infill angle 

The "fill angle in the print section -> infill -> advanced  section isn't enough?

> Are you talking about the "ironing" method?

I have to find the time to port it from my fork to your main branch. Shouldn't be that hard now.

Liger0, 2018-09-27:
> 
> 
> > top and bottom layers infill angle
> 
> The "fill angle in the print section -> infill -> advanced section isn't enough?
> 
> > Are you talking about the "ironing" method?
> 
> I have to find the time to port it from my fork to your main branch. Shouldn't be that hard now.

I mean I would like using different angle for the internal infill and the top and bottom layers.
Also using 360° (or more) as infill the infill will completely disappear , I think it is a king of bug, though using 0° should be the same mathematically wise, so at least it should be marked as an error to use a range between 0 and 359.

Liger0, 2018-09-27:
> I mean I would like using different angle for the internal infill and the top and bottom layers.
> Also using 360° (or more) as infill the infill will completely disappear , I think it is a king of [...]
With Cura I can specify a top layer coat over the normal top layers, Cura itself suggests only 1 is enough to have a smooth surface and in fact it is so, I don't know how that feature works though. I generally never used the ironing anyway.