slic3r/Slic3r #4548: [Request] Show the estimated time and money before exporting

Liger0, 2018-09-26:
### Version
1.3.1 -dev

### Operating system type + version
Windows 10 x64

### Feature request
It is pretty much simple. As of now the slicer only shows the amount of money, material and tiem after exporting to g-code. Since I mosty use that function to see how much I can play around with perimeters, infill and other settings to lower the cost and time, not needing to exprot every time may be good.

lordofhyphens, 2018-09-27:
The action that calculates the statistics is the gcode export.

Liger0, 2018-09-27:
> 
> 
> The action that calculates the statistics is the gcode export.

I see it actually is so, but it is really tricky in my opinion. In cura, for example, you don't need to export to gcode to see those statistics.