slic3r/Slic3r #4554: Occasionally Travel Moves are Generated With Extrusions.

ruevs, 2018-10-01:
### Version
Version 1.3.1-dev
Build commit: d17659b67c65bb5003ec92bf0d56ed7f8810e970
32bit
[Slic3r-master.2018.09.08.1848.d17659b.32bit.zip](https://dl.slic3r.org/dev/win/Slic3r-master.2018.09.08.1848.d17659b.32bit.zip)

### Operating system type + version
Windows XP SP3 32bit

### Wrong Behavior
Occasionally moves that should be travel moves are generated with extrusions.
Models similar to the attached one with "three spokes" occasionally show this on layers other than the first one as well.
[image]

#### STL/Config (.ZIP) where problem occurs
[Slic3rBug.zip](https://github.com/slic3r/Slic3r/files/2419890/Slic3rBug.zip)
[config.zip](https://github.com/slic3r/Slic3r/files/2419908/config.zip)

### Expected behavior
This is a regression of sorts. For example Slic3r-master.2018.04.26.1669.8a93fbb.32bit.zip (which I had laying around) does NOT have this problem when slicing the exact same object running on the same machine (so the exact same configuration as above).

[Slic3r-master.2018.04.26.1669.8a93fbb.32bit.gcode.zip](https://github.com/slic3r/Slic3r/files/2420018/Slic3r-master.2018.04.26.1669.8a93fbb.32bit.gcode.zip)

[image]
[image]

##### Initially I posted this here https://github.com/slic3r/Slic3r/issues/3755 but I suspect it is not the same issue so I am opening this one.

lordofhyphens, 2018-10-14:
So the problem goes away when you run the STL through netfabb. Joy. 

[Slic3rBug_fixed.zip](https://github.com/slic3r/Slic3r/files/2476836/Slic3rBug_fixed.zip)

lordofhyphens, 2018-10-17:
Service.netfabb.com is what I meant. 

I have not done a bisect yet ( it is a very long process to do so)

platsch, 2018-11-20:
I can confirm the bug, having the same issue with several models. The extrusions are actually printed and not only generated for the preview. Blaming the stl files to be inconsistent is a an easy, but not very helpful answer. I'm having this issue with a number of different files which are sliced correctly by older versions of Slic3r. Runnung every second model through netfabb is slightly annoying...

Unfortunately I have no clue of what could be the reason, didn't do substantial debugging though.

VanessaE, 2018-12-02:
I am experiencing this artifact, too (commit 1e4fb2d1b1811fcde5bbe82ae46c37302e0755ab).

Stl: [turn signal stalk repair.stl.zip](https://github.com/slic3r/Slic3r/files/2636476/turn.signal.stalk.repair.stl.zip)
Config: [config-20181201.ini.zip](https://github.com/slic3r/Slic3r/files/2636478/config-20181201.ini.zip)

lordofhyphens, 2018-12-02:
[reproduction_case.zip](https://github.com/slic3r/Slic3r/files/2636486/reproduction_case.zip)
Here's a cut-down version of @VanessaE model. 

Looks like it's in the thinwall code.

lordofhyphens, 2018-12-02:
The pseudocode for the regression test looks like thus:

Given: The 3D model [reproduction_case.zip] (https://github.com/slic3r/Slic3r/files/2636486/reproduction_case.zip) and a default config,
When: the model is plated centered at 0,0 and thin_walls=1

Then: No extrusion moves are generated from X1.856 Y-6.933 to X1.793 Y-6.899

supermerill, 2018-12-02:
My previous thin wall pr is seriously flawed (as said in the new one).

ruevs, 2019-04-18:
Any progress on this?
Slic3r-master.2019.04.13.2318.85dd6633.32bit-PR.zip
Still has exactly the same problem.

supermerill, 2019-04-18:
should be resolved on the branch merill_thinwall

ruevs, 2020-03-19:
It seem to be fixed in this AppVeyor build artifact:
https://ci.appveyor.com/project/lordofhyphens/slic3r/builds/31012289
from here:
https://github.com/slic3r/Slic3r/pull/4790