slic3r/Slic3r #4558: Dont show a grid

mrstanlez, 2018-10-04:
### Version
1.2.9

### Operating system type + version
Ubuntu 18.04 LTS fresh install at 4.october 2018

### Behavior
The problem is that Slic3r dont show a grid.
Reproduce the problem by starting a slic3r.

Expected Results is to show a grid

[image]

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
[config.zip](https://github.com/slic3r/Slic3r/files/2447206/config.zip)

lordofhyphens, 2018-10-04:
Use 1.3.0

lordofhyphens, 2018-10-14:
https://github.com/slic3r/Slic3r/releases/download/1.3.0/Slic3r-1.3.0-x86_64.AppImage

Does it still have a problem with that build? Will reopen if it does.

closed 2018-10-14 by lordofhyphens