slic3r/Slic3r #4560: Wide empty infill with concentric infill

Liger0, 2018-10-07:
### Version
1.3.1-dev

### Operating system type + version
Windows 10

### Behavior
Using concentric top and bottom layer infill, except for precise models done just considering everything like the line width, flow and so on (at least I guess), there will be widely empty parts on the top and bottom surfaces.
* _Expected Results_
I would think adding an automatic compensation for the gaps using different patterns would pretty much fix this and create a smooth surface. Even with a rectilinear infill there are a lot of gaps, but they are caused by the too small line width compared to the area that should be filled.

[image]

supermerill, 2018-10-08:
I also have this. I wonder if we should add gapfill lines to it ...

I can do that, it's easy but i need some input from other users on their opinions.

Liger0, 2018-10-08:
> 
> 
> I also have this. I wonder if we should add gapfill lines to it ...
> 
> I can do that, it's easy but i need some input from other users on their opinions.

It is important to say it is pretty much the same even using the rectilinear infill. I see the option to fill the gaps already is implemented with the rectilinear infill, but it really works bad, filling only a minimum amount of gaps and leaving about the 90%, especially the big ones.

supermerill, 2018-10-08:
>  I see the option to fill the gaps [...], but it really works bad, filling only a minimum amount of gaps and leaving about the 90%, especially the big ones.

hopefully my rework of thin wall will improve that as well.

> look at what happens everytime at the start and end point of each line

I don't understand. can you show me a layer where the outer perimeter is bad?
If you talk about the seam, there's nothing to do but to try to hide it in a corner.

Liger0, 2018-10-08:
> hopefully my rework of thin wall will improve that as well.
>
>
> I don't understand. can you show me a layer where the outer perimeter is bad?
> If you talk about the seam, there's nothing to do but to [...]

Isn't it possible to just continue the run more? Because it really doesn't touch the starting point. Also the seam options don't work, whatever I set they will always be in the same positions as in the image.

supermerill, 2018-10-08:
Somebody should merge this pr https://github.com/prusa3d/Slic3r/pull/728 then.

supermerill, 2018-12-10:
Done

[image]

but the amount of tiny fills can easily ruin a top surface: the gcode generator doesn't do a stellar job to order them (but it's at "good enough" level imo).
Should i force them to be lay down after the circles?
should i create a new infill (concentric+gapfill)?

Liger0, 2018-12-14:
What about only making it on the non-latest layers? So there won't be holes but still the externall skin won't be ruined. Also if ironing will be implemented it will be able to fill the skin gap too without ruining the face.

supermerill, 2018-12-14:
> Also if ironing will be implemented it will be able to fill the skin gap too without ruining the face.

I tried that. Not better than rectilinear.
Maybe worth a second try, it's easy to do.

Liger0, 2018-12-15:
> 
> 
> > Also if ironing will be implemented it will be able to fill the skin gap too without ruining the face.
> 
> I tried that. Not better than rectilinear.
> Maybe worth a second try, it's easy to do.

Or you can chose if to fill gaps normally like we said previously on the top surface too. Even if it will be messed, you can fix it ironing without extruding.

supermerill, 2018-12-15:
> you can fix it ironing without extruding.

i extrude with ironing, even cura extrude.
these gaps are too wide to be filled by an ironing, anyway.

supermerill, 2019-01-29:
cmmit for fix: supermerill/Slic3r@60442ba
may submit a pr when i have time.

tim-lukacik, 2020-10-20:
Any update on this? Other infills work just fine with these models
[image]
Cura produces the desired result
[image]

supermerill, 2020-10-29:
I'm not sure it's that desirable
[image]

Liger0, 2020-10-29:
> 
> 
> I'm not sure it's that desirable
> [image]

They are rewriting that gap filling for cura 4.9, you may take a look at their arachne implementation.