slic3r/Slic3r #4561: [request] Layer lines line compensation

Liger0, 2018-10-08:
### Version
v1.3.1-dev

### Operating system type + version
Windows 10

### Behavior
* _Expected Results_
On every line of each layer there should be an option to finish the travel when the starting point was, without leaving a gap everytime a new line is done
* _Actual Results_
There is a big gap on every line, it is very visible. The seam option doesn't even work, since on the same model changing the seam position will not do anything.

[image]

lordofhyphens, 2018-10-09:
So you want to deliberately slam the hotend into previously extruded material at the end of every layer?

Liger0, 2018-10-09:
> 
> 
> So you want to deliberately slam the hotend into previously extruded material at the end of every layer?

The gap is pretty large, I think some small movement before hitting the starting point could be done..

Bluelinegecko, 2018-11-06:
Yes, but is the gap a rendering error or a true slicing error? I suspect it is just rendering that gap and appearing way as the actual extrusion does not end in a perfect 90° edge in the center of the nozzle. If the gap is too large on a particular print of your have you calibrated your extruder or checked for backlash? I've never had an excessive gap in the seam in years of using slic3r,

bubnikv, 2018-11-06:
It is a real thing. The perimeter loop is split and a tiny gap is introduced to reduce the seam bulging out.

Liger0, 2018-11-06:
Yes, there is a huge "cut" on all the wall