slic3r/Slic3r #4562: [Feature Request] Add custom Start G-code field to Print Setting Tab

Bluelinegecko, 2018-10-12:
Currently there is a field for entering custom g-code at the start of a print in the filament and printer tabs With Marlin firmware's addition of linear advance, it would be handy to have a custom "Start G-code" field in the Print Setting tab as well. Having somewhere to specify the K factor for linear advance there would be appreciated as the nozzle pressure that affects the K-factor is more likely to be effected by layer height, print speed, extrusion width, etc.

Entering a specific K-factor in G-code in the printer or filament settings doesn't make as much sense. While those do affect the value one will often use the same filament/printer profiles with vastly different speeds and layer heights set in the print settings profile.

I'm sure other people would have other scenarios where they find it might be preferable to input their custom start g-code in the print settings as well.

lordofhyphens, 2018-10-12:
The main question I have is: Which custom gcode gets generated when?

Currently it is Printer -> filament 

My belief is that it should be printer -> filament -> print (and reverse order for end gcode)

lordofhyphens, 2018-10-12:
The use case allows people to put additional commands in filament settings (assume that the effects are undone in filament settings as well)