slic3r/Slic3r #4563: Missing/disappearing preset profiles for print settings

haleypearse, 2018-10-13:
### Version
slic3r PE 1.41.1-rc+
(3.4.1 hex file firmware MK3 variant with [#1193](https://github.com/prusa3d/Prusa-Firmware/pull/1193) merged)
https://github.com/prusa3d/Prusa-Firmware/files/2456589/Prusa-Firmware-3.4.1-fan-changes.zip

### Operating system type + version
Mac OS 10.14 Mojave

### Behavior

Every time I load slic3r or change printer profile I have to redo all my print settings, because the presets do not show. If I save a preset, it will show, but none others. Occasionally they will all come back temporarily. 

[image]

lordofhyphens, 2018-10-14:
https://github.com/slic3r/Slic3r/releases/download/1.3.0/slic3r-1.3.0.dmg

I will reopen if you can reproduce with that version.

closed 2018-10-14 by lordofhyphens

haleypearse, 2018-10-30:
Solved! It seems this was simpler than I'd thought. "Show incompatible print and filament presets" was unchecked, which seems to have been the problem. 
[image]

lordofhyphens, 2018-10-30:
I like it when it's a simple thing.