slic3r/Slic3r #4564: Slic3r 1.3 does not open on macOS 10.12.5

ganzocrypt, 2018-10-13:
Slic3r 1.3 does not open on macOS 10.12.5.

When launched it appears the window vault of verifying application and it sits there and does not go on.

lordofhyphens, 2018-10-14:
I wonder if Apple has decided to pull all opengl support officially from 10.12.5?

lordofhyphens, 2018-10-14:
I don't have 10.15 to test myself, unfortunately. 

If it's dying on the signed check that shouldn't be happening as @alexrj did get dev keys and the app should be signed. 

Could you try a build from https://dl.slic3r.org/dev/mac/ ?

ganzocrypt, 2018-10-14:
well, it's not dying, it just sits there.
I do not have the environment ready to do a build.
If no one reported must be something with my mac? not sure

lordofhyphens, 2018-10-15:
@ganzocrypt sorry, I just derped up and misread 10.12.5 as 10.15. AFAIK it should be working and this is the first report I've seen. I will test on my own system (on 10.14 High Sierra) tonight.

honza-kasik, 2018-10-17:
I have 10.13.16 and this appears when I try to open downloaded app: [image]

lordofhyphens, 2018-10-17:
@alexrj anything up with your key?

lordofhyphens, 2018-10-18:
hrm, OSX itself reports that it thinks the code has been modified from the signed version. That's weird to me because I don't recall messing with the build system. Now I just have to remember the URL for the build bot.

lordofhyphens, 2018-10-23:
So I've signed the dmg as well but forgot to do more testing. I'll get to it tonight. @ganzocrypt any improvement?

ganzocrypt, 2018-10-23:
I do not have an env to do the build so cannot help on that side, but willing to test the dmg as soon as you build it, just ping me.

lordofhyphens, 2018-10-23:
@ganzocrypt it should be up already. https://dl.slic3r.org/dev/mac/slic3r-1.3.1-dev-f487129.dmg

lordofhyphens, 2018-10-23:
@ganzocrypt you wouldn't be able to help with a build environment, unfortunately. It looks like it's Gatekeeper blocking execution.

ganzocrypt, 2018-10-23:
no luck.
it shows the window "Verifying Slic3r" .... and it sits there !

lordofhyphens, 2018-11-07:
Note: close this when #4576 closes.

lordofhyphens, 2018-11-10:
@ganzocrypt try it now.

closed 2018-11-10 by lordofhyphens

ganzocrypt, 2018-11-10:
which one?

lordofhyphens, 2018-11-10:
master-latest

ganzocrypt, 2018-11-10:
yes it works!
thx