slic3r/Slic3r #4566: Conducting research on Slic3r - Appreciate Help

JamesVerner, 2018-10-16:
Hey, not too sure if I should post an issue but I wasn't sure where I could go. However, what I'm doing currently is attempting to study the code if it was possible to modify existing patterns or implement a new structural support. Would anyone know which files I could see these two categories (for the structural support where existing ones are) and how they could be modified? I'm not looking to have a definitive answer, it's more the process and the train of thought. Thanks.

lordofhyphens, 2018-10-16:
SupportMaterial.pm for the (current) support generation code; it uses the patterns in Fill.hpp and its friends.

JamesVerner, 2018-10-17:
Thank you very much, will take a look now! Would SupportMaterial.ccp be the same, just the c++ format instead?

lordofhyphens, 2018-10-17:
It's someone's current port of it, and I'm pretty sure that it is unfinished (you are welcome to take a swing at finishing porting it from Perl).

JamesVerner, 2018-10-17:
Thanks for the offer but I doubt I will be able to help, still trying to understand the Fill.hpp you told me about. Just a quick question regarding the SupportMaterial.pm if you didn't mind answering? If I were to edit a pattern in, say fillRectangular, would SupportMaterial.pm need to be heavily adjusted as a result? Likewise, if I were to look into implementing a new structure, would SM.pm have to be adjusted to meet that as well?

lordofhyphens, 2018-10-17:
Fill material used is referenced at https://github.com/slic3r/Slic3r/blob/master/lib/Slic3r/Print/SupportMaterial.pm#L745

If you're using a new structure, you'll need to implement a new config option for it (https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.hpp#L36 , https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.hpp#L72) and wire it in (https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/Fill/Fill.cpp#L21). 

Your new pattern needs to implement Fill. 
https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/Fill/Fill.hpp#L20

You can play around with changing the used fill type to other kinds to see if that does anything interesting for you (like the 3d fills).

JamesVerner, 2018-10-17:
You're a Saint, thank you again!

lordofhyphens, 2018-10-17:
SupportMaterial.pm implements pillar support (its own thing) and then a solid-style support option. It's the solid-style support that uses normal infill (so far as I am aware). 

It's also the area doing the detection on what needs to be supported (which surfaces, etc) and the interface patterns. 

FYI, if you've made the pattern available via making the additions to PrintConfig and Fill.cpp, you can select it as a regular *infill* pattern as well.

lordofhyphens, 2018-10-17:
The way that Infill patterns work is that they usually scale some geometric shape out and repeat it. 

In about 20-40 minutes, you should be able to see the fill.hpp doxygen docs fill out over at http://manual.slic3r.org/libslic3r-doc/class_slic3r_1_1_fill.html

JamesVerner, 2018-10-17:
Hahaha thank you again for replying to everything, you've helped me to get a grip roughly how it works. Likewise don't worry I won't be pushing anything that will break the project. Sorry, just for clarification. If I was to create a new structure, I would just need to reference it in the SupportMaterial.pm and then regarding Fill. I would just follow the structure as shown in http://manual.slic3r.org/libslic3r-doc/class_slic3r_1_1_fill.html? So FillNewStructure.hpp, .cpp, etc?

lordofhyphens, 2018-10-17:
You'll need to add it to the enumeration and keymap in PrintConfig.hpp, return an instance of it in https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/Fill/Fill.cpp#L21, and then you can change the fill as its called by name in SupportMaterial.hpp (you can play around by changing the name referenced to patterns that already exist). 

I think you can get it in the dropdown for general use (there's a pattern option) 
by adding it to https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.hpp#L90 and https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.hpp#L43

And then, yes, FillNewStructure.hpp/cpp, etc.

JamesVerner, 2018-10-17:
Thank you again for all your help, hopefully, I get something out of all this.

JamesVerner, 2018-10-18:
Hey, sorry just to wrap this up and if you had to the time. I just wanted to take your information, go to the classes and then try and follow the process so in order to implement a new structural support (rectilinear, honeycomb, rectilinear grid, pillars). To show in the drop down list for 'Print Settings', we add the smpNewStructure and key_map... to:(https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.hpp#L90)
(https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.hpp#L43)

Which then would be referenced somewhere in the supportMaterial.pm? Afterwards we need to begin the infillPattern stage for this support structure (NewStructure). With that we add the same logic as before, in:
(https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.hpp#L36) (https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.hpp#L72)

Then as you said, wire it in the switch:
(https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/Fill/Fill.cpp#L21)

After that we follow the heiarchy structure of Fill.hpp, so FillNewStructurePattern.hpp, FillNewStructurePattern.cpp? 

Is that then the process of adding a new structural support? Have I miss anything or understood anything wrong? Thank you again.

lordofhyphens, 2018-10-23:
Sorry for taking forever to get back to you. 

Yes, it looks like you have the right of it. 

If you want to remove the "walls", I believe that is around https://github.com/slic3r/Slic3r/blob/master/lib/Slic3r/Print/SupportMaterial.pm#L842

JamesVerner, 2018-10-23:
No worries, thank you for double checking and replying back. Just one last thing, I wanted to reference a new dummy value in the drop-down list shown here:
https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.hpp#L90
https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.hpp#L43
https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.cpp#L1516
So effectively what I did was...     keys_map["pillars"]             = smpNewStructure;

However it didn't seem to generate the support, is there something I am missing? I noticed there is a switch statement in the Fill.cpp so I thought it might be the same and have spent a day looking for it but no success.

lordofhyphens, 2018-10-26:
I think all of the code that makes it a pillar is on the perl side, but I haven't looked at that code in a while.

JamesVerner, 2018-10-26:
Hey, thanks again. Take it that would be just the Supportmaterial.pm one
you mentioned?

lordofhyphens, 2018-10-27:
Yes.

JamesVerner, 2018-10-28:
Thanks, will stop harassing the issue tracker. I don't suppose (https://github.com/slic3r/Slic3r/blob/44d145d986c7bf5124910de1b561836071bf8305/lib/Slic3r/Print/SupportMaterial.pm#L72) Would allow me to edit the size of the column, both the top and bottom? Such as if I wanted to have a thicker bottom but a thinner top?

closed 2018-11-06 by lordofhyphens