slic3r/Slic3r #4567: OpenGL-0.70 config failure

Kuro-Maii, 2018-10-18:
### Version
```
$ git rev-parse --verify HEAD
f487129c7059e72b07a8179f1b8968b3a361739c
$ git describe --tag
1.3.0-1007-gf487129c
```
slic3r > about
1.3.1-dev

### Operating system type + version
```
$ uname -a
Linux Yggdrasil 4.9.0-8-amd64 #1 SMP Debian 4.9.110-3+deb9u6 (2018-10-08) x86_64 GNU/Linux
$ cat /etc/debian_version
9.5
$ lspci | grep -i 'vga'
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Hawaii PRO [Radeon R9 290/390]
```
all drivers are up-to-date

### Behavior
* _Describe the problem_: Configuration failure of the required OpenGL libraries
* _Steps needed to reproduce the problem_: succesfully build with `perl Build.PL` then run `perl Build.PL --gui` like instructed
* _Expected Results_:
```
$ perl Build.PL --gui --verbose
App::cpanminus is up to date. (1.7044)
Class::Accessor is up to date. (0.51)
Growl::GNTP is up to date. (0.21)
LWP::UserAgent is up to date. (6.36)
Net::Bonjour is up to date. (0.96)
--> Working on OpenGL
Fetching http://www.cpan.org/authors/id/C/CH/CHM/OpenGL-0.70.tar.gz ... OK
Configuring OpenGL-0.70 ... SUCCES
# rest of successful compilation/configuration
$ 
```
* _Actual Results_:
```
 perl Build.PL --gui --verbose
App::cpanminus is up to date. (1.7044)
Class::Accessor is up to date. (0.51)
Growl::GNTP is up to date. (0.21)
LWP::UserAgent is up to date. (6.36)
Net::Bonjour is up to date. (0.96)
--> Working on OpenGL
Fetching http://www.cpan.org/authors/id/C/CH/CHM/OpenGL-0.70.tar.gz ... OK
Configuring OpenGL-0.70 ... FAIL
! Timed out (> 60s). Use --verbose to retry.
[1]    17679 terminated  perl Build.PL --gui --verbose
! Configure failed for OpenGL-0.70. See /home/angel/.cpanm/work/1539854824.17707/build.log for details.                                  

 $ pastebinit /home/angel/.cpanm/work/1539854824.17707/build.log
```
[paste of log](http://paste.debian.net/1047872/)

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
N/A

lordofhyphens, 2018-10-26:
What version of wxWidgets is installed by default in your distro?

lordofhyphens, 2018-10-26:
@Kuro-Maii you did have opengl headers available on your system, right? Mostly trying to cover most of the situations where POGL would fail to config/build. 

Also, your pastebin is gone now. You can drop txt files right onto github, best to just do that.

Kuro-Maii, 2018-11-12:
Sorry for the delay...
I did a fresh pull and build today
the version I got for wxWdigets
```
$ wx-config --version
3.0.2
```
I can't find specific instructions to show if I got the opengl headers avalible, so the best I can do right now is:
```
$ glxinfo | grep "OpenGL"
OpenGL vendor string: X.Org
OpenGL renderer string: Gallium 0.4 on AMD HAWAII (DRM 2.49.0 / 4.9.0-8-amd64, LLVM 3.9.1)
OpenGL core profile version string: 4.3 (Core Profile) Mesa 13.0.6
OpenGL core profile shading language version string: 4.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 13.0.6
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.1 Mesa 13.0.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
OpenGL ES profile extensions:

$ ll /usr/include/GL
total 1,5M
-rw-r--r-- 1 root root 8,8K aug 23  2016 freeglut_ext.h
-rw-r--r-- 1 root root  681 aug 23  2016 freeglut.h
-rw-r--r-- 1 root root  26K aug 23  2016 freeglut_std.h
-rw-r--r-- 1 root root 236K apr 25  2017 glcorearb.h
-rw-r--r-- 1 root root 783K apr 25  2017 glext.h
-rw-r--r-- 1 root root  79K apr 25  2017 gl.h
-rw-r--r-- 1 root root 156K apr 25  2017 gl_mangle.h
-rw-r--r-- 1 root root  17K sep 13  2015 glu.h
-rw-r--r-- 1 root root 3,3K sep 13  2015 glu_mangle.h
-rw-r--r-- 1 root root  639 aug 23  2016 glut.h
-rw-r--r-- 1 root root  47K apr 25  2017 glxext.h
-rw-r--r-- 1 root root  15K apr 25  2017 glx.h
-rw-r--r-- 1 root root 4,6K dec 11  2013 glxint.h
-rw-r--r-- 1 root root 3,4K apr 25  2017 glx_mangle.h
-rw-r--r-- 1 root root 2,1K dec 11  2013 glxmd.h
-rw-r--r-- 1 root root  80K dec 11  2013 glxproto.h
-rw-r--r-- 1 root root  12K dec 11  2013 glxtokens.h
drwxr-xr-x 2 root root 4,0K apr 21  2018 internal
```

so my current slic3r tree is at:
```
$ git describe --tag
1.3.0-1080-g1e4fb2d1
$ git rev-parse --verify HEAD
1e4fb2d1b1811fcde5bbe82ae46c37302e0755ab
```

`perl Build.PL` still results in a successful compile

```
$ perl Build.PL --gui --verbose
App::cpanminus is up to date. (1.7044)
Class::Accessor is up to date. (0.51)
Growl::GNTP is up to date. (0.21)
LWP::UserAgent is up to date. (6.36)
Net::Bonjour is up to date. (0.96)
--> Working on OpenGL
Fetching http://www.cpan.org/authors/id/C/CH/CHM/OpenGL-0.70.tar.gz ... OK
Configuring OpenGL-0.70 ... FAIL
! Timed out (> 60s). Use --verbose to retry.
[1]    15010 terminated  perl Build.PL --gui --verbose
! Configure failed for OpenGL-0.70. See /home/angel/.cpanm/work/1542029509.15042/build.log for details.
```
results in `SIGTERM(15)` after 00:01:02
[build.log](https://github.com/slic3r/Slic3r/files/2572083/build.log)

I learned of a workaround by hitting `[ctrl]+c`
which resulted in a successful compilation with:
```
$ perl Build.PL --gui -- verbose
App::cpanminus is up to date. (1.7044)
Class::Accessor is up to date. (0.51)
Growl::GNTP is up to date. (0.21)
LWP::UserAgent is up to date. (6.36)
Net::Bonjour is up to date. (0.96)
--> Working on OpenGL
Fetching http://www.cpan.org/authors/id/C/CH/CHM/OpenGL-0.70.tar.gz ... OK
Configuring OpenGL-0.70 ... ^CDon't worry, this module is optional.
--> Working on Socket
Fetching http://www.cpan.org/authors/id/P/PE/PEVANS/Socket-2.027.tar.gz ... OK
Configuring Socket-2.027 ... OK
Building and testing Socket-2.027 ... OK
Successfully installed Socket-2.027 (upgraded from 2.020_03)
1 distribution installed
--> Working on Wx
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Wx-0.9932.tar.gz ... OK
Configuring Wx-0.9932 ... OK
Building and testing Wx-0.9932 ... OK
Successfully installed Wx-0.9932 (upgraded from 0.9928)
1 distribution installed
Wx::GLCanvas is up to date. (0.09)
```

lordofhyphens, 2018-11-12:
Whatever is going on, it is specific to your computer's drivers, libraries, etc.

lordofhyphens, 2018-11-12:
It is required for the plater GUI. You can still use the command line interface without it.

You can also install it by hand or from your distro packages.
https://github.com/KohaAloha/Perl-OpenGL/blob/master/INSTALL

Once you have run build.PL --gui once, you don't need to run it with the --gui flag again on that install of Slic3r. All the --gui flag adds is installation of dependencies for gui stuff.

Kuro-Maii, 2018-11-12:
so.. thanks to the fact that I ran ubuntu before and have the git repo for slic3r in my home folder made it possible to run the slic3r with a gui, without problems, when I switched to debian at the start of this year.

regarding installing the distro packages. everything is installed with the exception of your linked repo from its source. due to reservations with 8 year old repos.

lordofhyphens, 2018-11-12:
@Kuro-Maii then you're probably fine anyway and should never need to run build.pl --gui. 

What I can tell you is that I can build it fine on my debian system. There's not much more I can do (because it's not a bug in Slic3r) and if you want to get it to build automatically you should go over to the POGL repo (http://sourceforge.net/projects/pogl).

closed 2018-11-12 by lordofhyphens

Kuro-Maii, 2018-11-12:
Thank you for helping and looking into my issue. :+1: