slic3r/Slic3r #4569: Placeholder not working?

hoh61, 2018-10-28:
**Describe the bug**
For a filament specific command I need to know to which extruder this filament is assigned. Unfortunately I failed up to now to find the suitable placeholder. 

From the list in the cartesio wiki I assumed that the "filament_settings_id" could be a suitable placeholder, but this placeholder is delivering nothing.

Is there any placeholder I can use for this?

; Filament gcode
M214 T[filament_settings_id'] ...

**Desktop (please complete the following information):**
 - Linux
 - Version 4.4.0-66-generic
 - generated by Slic3r 1.3.1-dev (Build )

lordofhyphens, 2018-10-30:
Can you upload a CTRL-E export so I can verify this with your config? Thank you.

lordofhyphens, 2018-10-30:
Which custom gcode are you trying to use this with? There's a few that I suspect aren't valid in Start or End Gcode, but would be valid in before/after layer gcode or the toolchange gcode.

lordofhyphens, 2018-10-30:
What I have been able to determine is that it's a "legit" Slic3r setting, but it's being set to "". That's also what it is in the config. 
````
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
Placeholder: DEBUG: Replacing filament_settings_id with ().
````

I have no idea at present if this was originally going to be automagically set or it was intended to be a configuration value. 

I'm going to give it a name and tooltip so you can mark filaments in this manner.

lordofhyphens, 2018-10-30:
You can now set the option as of https://github.com/slic3r/Slic3r/commit/d9c1d459c5744aa880cd0c5e9093b0ee42f5321b. It should interact just fine with custom gcode, as well.

lordofhyphens, 2018-10-30:
Yeah, it looks like @alranel added it as an option at some point but never wired it into anything (nothing sets this value as far as I can tell with grep).

This seems like a good use for it.

lordofhyphens, 2018-10-30:
Reopened as @hoh61 has indicated that this is not what they were looking for :)

hoh61, 2018-10-31:
A try for clarification:

In the plater GUI you are assigning the filament to a specific nozzle. Based on this setting the material specific parameters will be written in the gcode with  proper tool settings:

e.g. Filament with extrusion temperature of 225°C to extruder 2:
-> M109 S225 T1

In the same way I need the tooling index as a parameter for a material specific command, which is located in the filament specific code option and shall set some extrusion parameters in the firmware, in my case the pressure advance parameters. So my Idea was to have the following code in the filament specific gcode:
M214 T[filament_settings_id] K15 D25.

For the filament above, which is assigned to the second extruder, it shall translate in the gcode to 
-> M214 T1 K15 D25

By this way I would like to ensure a proper parameter setting independent from the actually used position. Since I have different material with significant variations this would save some bad prints.

I don't know if you have such a variable available externally, however , the naming of my selected parameter looked encouraging.  I hope, there can be an easy solution soon.

lordofhyphens, 2018-11-06:
Right now, [filament_settings_id] is a user-set parameter that is whatever you tell it to be.

I wonder how much of a pain it would be to have something like [this_extruder] (I can't remember offhand if we have a general reference to the current extruder that exists outside of the toolchange gcode) exist in the general case and overload in the case of the filament settings to always resolve to the extruder where the filament lives.

lordofhyphens, 2018-11-06:
@hoh61 So you want a replacement ID that, in filament_gcode for start and end gcode, refers to whatever position (0-N) that filament lives in.

lordofhyphens, 2018-11-06:
@hoh61 I've opened a pull request; is that what you're looking for? 

You can test it out by downloading [GCode.zip](https://github.com/slic3r/Slic3r/files/2551320/GCode.zip) and extracting it to slic3r/lib/Slic3r/Print. Or checking out the pull request (either/or).

lordofhyphens, 2018-11-06:
@hoh61 https://manual.slic3r.org/advanced/placeholder-parser as a FYI (it's new)

lordofhyphens, 2018-11-06:
@hoh61 Yeah, I had a typo that I've fixed in the related pull request.

lordofhyphens, 2018-11-06:
Closing issue as related PR has been merged. Next automatic build should have the changes.

closed 2018-11-06 by lordofhyphens