slic3r/Slic3r #4570: Incapable of slicing properly for thin models.

TaiTair, 2018-10-29:
**Describe the bug**
Slic3r incapable of filling properly with thin models.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the stl provided in this report.
2. Slice it with nozzle width of 0.4mm, layer height of 0.15mm and first layer height of 0.20mm. 
2 Perimeters and 3 solid layers.  Infill at 20% rectilinear.
Use first layer extrusion width of 200%, top solid infill of 150% and infill/perimeters overlap of 30%.
All other settings not important for this test.
3. See how the model slices incorrectly.

**3D Model and Slic3r Configuration Export**
[bugresources.zip](https://github.com/slic3r/Slic3r/files/2527006/bugresources.zip)

**Expected behavior**
I expected it to fill properly without any gaps.

**Screenshots**
Noticeable gaps in infill, straight lines and not enough layers:
[image]
In comparison, sliced with curaengine, it looks how it should look:
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - Version 1.2.9

supermerill, 2018-10-29:
the  1.2.9 is 3 years old.
at least, can you try with the 1.3? It's only 5 month old.

also

> I expected it to fill properly

It's a bit light as a description. Point the problematic area in the picture, and explain why it's bad.

TaiTair, 2018-10-29:
> the 1.2.9 is 3 years old.
> at least, can you try with the 1.3? It's only 5 month old.
> 
> also
> 
> > I expected it to fill properly
> 
> It's a bit light as a description. Point the problematic area in the picture, and explain why it's bad.

Are you blind or are you just trying to be annoying? Use your eyes, the gaps in the infill are clearly visible without even zooming in the pictures. Also the infill is a lot lighter and not a good representation of the model, whereas cura makes an accurate representation of the model. Slic3r makes an incomplete, flimsy model.

Here I downloaded this one: https://dl.slic3r.org/dev/linux/Slic3r-master-latest.tar.bz2 and it wouldn't even launch due to perl errors and a misconfiguration of local libs.

[image]

I don't have the time to test the release packages and make them work. If I wanted to compile I'd grab the source. When I download a pre-compiled "release" package I expect it to work without confusion.
Wish I could test the 1.3.

lordofhyphens, 2018-10-30:
@TaiTair Use the AppImage.

Which of the settings from the bundle were you testing with (the Ctrl-E export is asked for to avoid this question).

lordofhyphens, 2018-10-30:
I took a guess from the bundle (all three "My Settings.ini") and produced [4570.zip](https://github.com/slic3r/Slic3r/files/2527514/4570.zip)
Here's the slice from the top with 1.3.1-dev:
[image]

I originally thought you were complaining about the tail section, so I had zoomed in on those. 

0.65:
[image]

1.25:
[image]

3.05:
[image]

lordofhyphens, 2018-10-30:
@TaiTair Odd, on my Debian Sid system the tar.bz2 starts right up when calling ./Slic3r. 

Something's up with the AppImage on my system (that I just noticed) when downloading it as a check. Looking into it now.

TaiTair, 2018-10-30:
@lordofhyphens Thanks for your help.
I tested the AppImage and the result was much better - comparable to cura's result:
[image]

So the incorrect slicing seems to be solved with the latest versions. Now the next question is how do I get the tar.bz2 to work? I'd like to embed it in my distro so it can be called from repetier-host. Going to try building it from source.

TaiTair, 2018-10-30:
Ok I managed to compile 1.3.1-dev without issues and to use it.
The problem is mostly gone and "Detect thin walls" works better. I tweaked it a bit and managed to get rid of gaps on the top of the model.
That being said I still got a very annoying gap under the model. Which settings should I try changing? What am I doing wrong?

[image]

[config.txt](https://github.com/slic3r/Slic3r/files/2530440/config.txt)

TaiTair, 2018-10-30:
First layer view, very noticeable gap:
[image]

TaiTair, 2018-10-30:
Yeah... No matter what I do the sliced model has gaps.

TaiTair, 2018-10-30:
Looks like the logic for slic3r is flawed. The paths in Cura look way more logical. Slic3r on the left, cura on the right. Layer height settings are exactly the same.
[image]

TaiTair, 2018-10-30:
@lordofhyphens Could you look at your first layer and check for any gaps?
Even with the AppImage I get gaps in my slices - whereas yours looks perfect. I have a feeling maybe some of my dependencies are glitching out?

lordofhyphens, 2018-10-30:
@TaiTair Yeah, I can take a look at it when I get some time tonight. I had used your three profiles labeled "My Settings" to generate my own screencaps. 

Using Slic3r's own preview with the AppImage would give what Slic3r thinks the paths look like. You also can take a look at Slic3r's own host mode :)

With the tar.bz2, you'd tell rephost to call the Slic3r script that's in the dir.

TaiTair, 2018-10-31:
Yeah so I tried running it on my Windows partition and the slicing was pretty much flawless. No gaps or anything. Which tells me something might be wrong with my Ubuntu.

Very interesting. So I tried slicing it directly from slic3r and looking at the preview inside of slic3r's gui and it seemed flawless too:

[image]

Now I'm at a loss because it seems like the problem is inside Repetier-host. I'm pretty sure i've seen those gaps in my prints. Is it possible repetier is messing things up?
Loaded the g-code from slic3r into repetier again (sliced directly with slic3r and then launched repetier separately)
Again some gaps on almost every layer.

[image]

What's going on??

lordofhyphens, 2018-10-31:
Have you tried to print it?

TaiTair, 2018-10-31:
Yeah gcode.ws not showing me any gaps with this:

[F-47 Mini .gcode.txt](https://github.com/slic3r/Slic3r/files/2535828/F-47.Mini.gcode.txt)

Had to .txt it because Github doesn't take .gcode for attachements.

lordofhyphens, 2018-10-31:
@TaiTair it does take .zip though :) 

If Gcode.ws doesn't think there is something missing, and Slic3r doesn't think there should be something missing, then I'd try to print it.

TaiTair, 2018-10-31:
Alright I'll try soon as I get back from work.

TaiTair, 2018-10-31:
Little bit of digging around:

Seems like the cause for my problem is actually Repetier-host and I have the feeling it actually prints the problem (I'll double check tonight):
https://forum.repetier.com/discussion/4905/gaps-between-layers-in-print-preview

Here's a preference menu inside Repetier that needs to be setup for previews to work correctly. I have yet to set it up properly.

[image]

lordofhyphens, 2018-11-01:
Yeah, I am not a user of RepetierHost and I will likely never be one.

If Gcode.ws and Slic3r both generally agree that there is no major issue, I am comfortable closing this.

I suspect that Printrun would also show all is well.

If the gcode output from Slic3r is missing those areas then I will reopen.

closed 2018-11-01 by lordofhyphens

TaiTair, 2018-11-01:
I was just about to close it but you beat me to it. The print was flawless and the problem comes from RepetierHost's preview system. 

Thanks for taking the time, cheers.