slic3r/Slic3r #4571: Slic3r under certain conditions when loading external configs

lordofhyphens, 2018-10-30:
**Describe the bug**
Slic3r crashes when loading filament_settings_id option under some circumstances, such as provided.

**To Reproduce**
Steps to reproduce the behavior:

1. Download [config.zip](https://github.com/slic3r/Slic3r/files/2527790/config.zip)
1. Extract config.ini
1. Go to File->Load Config
1. Select config.ini
4. Watch it crash with a segfault.

**3D Model and Slic3r Configuration Export**
[config.zip](https://github.com/slic3r/Slic3r/files/2527790/config.zip)

**Expected behavior**
Slic3r does not crash on loading config.

**Desktop (please complete the following information):**
 - OS: Linux
 - Version 1.3.1-dev d9c1d459c5744aa880cd0c5e9093b0ee42f5321b
     - _Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._
     - _For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     - _Use `About->About Slic3r` for release versions._

**Additional context**
Workaround is to manually edit the configuration to avoid the bug state. filament_settings_id is not being loaded from defaults appropriately when going from a single extruder to multiple extruder. 

Add `filament_settings_id = ` with N-1 semicolons, where N is the number of extruders.

Abbreviated traceback from coredump:
````
#0  0x00007f2600775b06 in Slic3r::ConfigBase__set(Slic3r::ConfigBase*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, sv*) (THIS=<optimized out>, opt_key=..., value=<optimized out>)
    at /usr/include/c++/8/ext/new_allocator.h:79
#1  0x00007f26006ad428 in XS_Slic3r__Config_set(interpreter*, cv*) (my_perl=<optimized out>, cv=<optimized out>) at /usr/bin/perl -MExtUtils::XSpp::Cmd -e xspp -- -t "../xsp/typemap.xspt"  "../xsp/Config.xsp":83
#2  0x000055985c6c9741 in Perl_pp_entersub ()
#3  0x000055985c6c1676 in Perl_runops_standard ()

````

gege2b, 2018-11-03:
No crash at all on my linux mint, not even a warning or something similar, settings seems to be properly loaded

lordofhyphens, 2018-11-06:
#4577 reverts the change to the config system.

lordofhyphens, 2018-11-06:
Going to close as I reverted the related change.

closed 2018-11-06 by lordofhyphens