slic3r/Slic3r #4576: Mac builds are not correctly signed

alranel, 2018-11-05:
As of 4df8935, dev builds are still not correctly signed:

```
$ codesign --verify /Volumes/Slic3r/Slic3r.app
/Volumes/Slic3r/Slic3r.app: CSSMERR_TP_NOT_TRUSTED
```

lordofhyphens, 2018-11-06:
I have an Apple dev account as well, I've been meaning to figure out how the code signing system works as well so that my keys can be available as a backup.

lordofhyphens, 2018-11-06:
From IRC: Looks like it's a function of the age of the OS on the build server (10.9). Will upgrade to 10.12 tonight.

lordofhyphens, 2018-11-07:
Should be sorted out now; current latest master now opens on my 10.14 box.

lordofhyphens, 2018-11-07:
Jenkins is back up. Here's the error from the log 
````
Signing app...
./package/osx/Slic3r.app: A timestamp was expected but was not found.
In subcomponent: /Users/Shared/Jenkins/Home/workspace/Slic3r/package/osx/Slic3r.app/Contents/MacOS/local-lib/lib/perl5/Algorithm/Diff.pm
````

lordofhyphens, 2018-11-09:
https://dl.slic3r.org/dev/mac/slic3r-1.3.1-dev-4ae028a7.dmg Looks like it's signing properly now.

lordofhyphens, 2018-11-10:
Closing; build server is now signing correctly and it looks like it's verifying on my own system.

closed 2018-11-10 by lordofhyphens