slic3r/Slic3r #4578: saving picture for every layer

rafaa1988, 2018-11-06:
Hi all,
Slic3r slices an stl model into layers and then convert the stl model to gcode.
Is it possible to save .jpg picture for every sliced layer?

Thanks

lordofhyphens, 2018-11-06:
@rafaa1988 there's SVG export already.

rafaa1988, 2018-11-06:
I tried file -> slice to svg 
but i get only one svg file and it's blank.
I'm expecting to have many svg files for all sliced layers. in each it shows the tool path.

lordofhyphens, 2018-11-06:
The SVG file isn't actually blank, the color of the lines are white. I believe that this is an artifact of its original use: SVG output was primarily used for DLP/SLA applications.

alranel, 2018-11-21:
The _Slice to SVG_ command in the File menu exports solid slices.
The _Export SVG_ command in the DLP projector dialog exports SLA toolpaths (infill patterns etc.)
There's no way to export SVG files or bitmaps representing the FDM G-code toolpaths.