slic3r/Slic3r #4587: Adaptive Slicing - visual quality with tall thin segments or rounded hallow domes

storagejarett, 2018-11-10:
**Describe the bug**
(Posting on LoH's request)

Adaptive slicing may not slow down print speed on thin tall objects but rather just smoosh filament into place (vs s3d which slows down and does a small circle).  Also, thin walled hallow domes look very jagged. 

**To Reproduce**
Steps to reproduce the behavior:

Printed Ctl-V v3 same test with all settings besides adaptive slicing the same as S3D.  

**3D Model and Slic3r Configuration Export**

https://www.thingiverse.com/thing:1363023

**Expected behavior**

Cleaner pillars and domes
[testprintandgode.zip](https://github.com/slic3r/Slic3r/files/2568684/testprintandgode.zip)

**Screenshots**
Multiple comparison screenshots here.  
https://imgur.com/a/u2UyeOB

**Desktop (please complete the following information):**
 - OS: OSX Mojave 10.14
 - Version  1.3.1-dev 

**Additional context**
Add any other context about the problem here.

lordofhyphens, 2018-11-12:
@platsch

platsch, 2018-11-12:
I don't exactly understand the request. What is the behavior you observer for tall objects and what do you suggest to improve it?
As for the hollow dome: I suspect it is an effect caused by the negative slope at the inside of the dome, didn't do a closer investigation though. That's on the list for quite a long time now, shouldn't be so hard to fix actually, but I just don't find the time. Would be happy to support anyone who would give it a shot.