slic3r/Slic3r #4594: README.md is misleading

albertvanderhorst, 2018-11-12:
**Describe the bug**
The first thing you read is that there are Mac and Windows development possible. This suggest that you can't develop on Linux, which is a totally wrong first impression. 

**Expected behavior**
The first impression should be that main development is done on linux.

**Additional context**
A solution is to change the text:

Beside linux 32 and 64 bit development systems we have not also ...

Note: it may be that linux 32 development is not possible, this is why this first line anybody new sees is so important.

lordofhyphens, 2018-11-12:
Make the edit you think it should be and open a pull request ;)

albertvanderhorst, 2018-11-29:
I would have, but if you look closer you see that I don't have theinformation. You do not suppose I clone the archive, then find out whether 32 bit development is possible, then change one line, while you know that out of the top of your head.

Beside linux 32 and 64 bit development systems we have not also

Beside linux 64 bit development systems we have not also

You merely have to paste one of the above phrases, whatever is appropriate.