slic3r/Slic3r #4597: Not Drawing bottom layer

inventaholic, 2018-11-16:
### Version
_Version of Slic3r used goes here_
PE versino 1.41.1

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
Not drawing bottom layer for some parts. this has come up a handful of times for me. STL attached. 
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com
[image]

/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2018-11-17:
Show that this is a problem on a build from dl.slic3r.org/dev/ or https://dl.slic3r.org/ and I will reopen.

closed 2018-11-17 by lordofhyphens

lordofhyphens, 2018-11-17:
[image]
Can't reproduce with my settings either, so if this is a bug in Slic3r it's related to your settings. 

If you can reproduce with Slic3r (not Slic3r PE), add a config export to your issue (File->Export Config). 

If you can't reproduce with Slic3r, the PE issue tracker is https://github.com/prusa3d/slic3r/issues