slic3r/Slic3r #4602: Slicing From Command Prompt Not Working

samlochner, 2018-11-20:
Following the instructions here: https://manual.slic3r.org/advanced/command-line

When I try to slice a model from the command prompt: slic3r my_model.stl --layer-height 0.2

Slic3r opens and the model is loaded, but no slicing happens. Anyone have any idea what the issue might be?

Also, I have some general questions about scripting slic3r. Is this the right place to ask them? If not, any suggestions on a more appropriate place?

Thank you kindly,
Sam

lordofhyphens, 2018-11-20:
Yes, the issue tracker is fine.

samlochner, 2018-11-20:
Okay, I'll have multiple questions. First thing though is why the command "slic3r my_model.stl --layer-height 0.2 " only opens slic3r and the model but doesn't create the gcode.

Thanks,
Sam

lordofhyphens, 2018-11-20:
Separate questions should go in separate issues.

lordofhyphens, 2018-11-20:
What version of Slic3r are you using?

samlochner, 2018-11-20:
Okay,

I'm using Prusa Edition 1.4

samlochner, 2018-11-20:
I just tried with upstream Slic3r and still the same issue: Sli3er opens and the stl file opens but no g code is generated. Here's the script again I'm running as per the manual:

slic3r my_model.stl --layer-height 0.2

Any idea what might be going on?

Thanks,
Sam

alranel, 2018-11-21:
@samlochner are you using Windows?

samlochner, 2018-11-21:
Yes I am, and you asking that helped me see my mistake. I was not using the console version. Thank you!

I'm just getting started with automating here, and I appreciate the patience :-)
Sam

closed 2018-11-21 by samlochner