slic3r/Slic3r #4603: Question: Access modifiers from command prompt?

samlochner, 2018-11-21:
I'm hoping to slice from the command prompt while using a modifier stl to make part of my model with 100% infill. Is there a command to do this? 

In general, is there anywhere I can see a complete list of all the commands available?

Thanks,
Sam

supermerill, 2018-11-21:
list for all options : https://github.com/slic3r/Slic3r/blob/4fcbebf0ff9b29683da7262d51e9fb7d78ddfeb2/xs/src/libslic3r/PrintConfig.cpp

look at each line with "cli" (you can "grep cli" to have only the option, but i think the description may be useful)

lordofhyphens, 2018-11-21:
I don't believe you can use modifier meshes from the CLI.

samlochner, 2018-11-21:
Shoot... Can anyone think of an alternative approach to specify a region of 100% infill from the command prompt, like maybe two overlapping models with different infills?

Is there someone out there who could get modifiers working in the CLI?

Thanks,
Sam

lordofhyphens, 2018-11-22:
You could probably set it up with the GUI ahead of time and save the plate. No idea if that will actually work.

There is a lot of UI complexity when it comes to modifier meshes

samlochner, 2018-11-22:
bubnikv, what goes in the XML files? That can automate modifiers? Do you have any examples?

lordofhyphens, you mean arrange the files in a CAD program before importing? The solid parts being used instead of modifiers would still be arranged automatically away from their original position. And I don't see how I can use "--merge" while maintaining different infills... 

Being able to specify different infill per region seems like a fairly necessary task, there's got to be a way to automate this...

Thanks,
Sam

bubnikv, 2018-11-22:
Just export the files from the UI and analyze them.

samlochner, 2018-11-22:
Brilliant! Thanks!!

Sam

closed 2018-11-22 by samlochner