slic3r/Slic3r #4605: Auto Arrange of sequential objects sees edge of bed as an object

Mirarkitty, 2018-11-23:
**Describe the bug**
When auto arranging sequential objects, with a setting of a print head radius, the objects will never be closer to the print bed edge than this radius. This lowers the amount of objects that can be on the print bed.

**To Reproduce**
Steps to reproduce the behavior:
1. load a small object
2. set sequential printing and printer head radius
3. increase the number of objects until they just fit on the print bed
4. increase the radius of the printer head until they don't fit on the bed
4. notice that the radius from any object to the print bed is always more than the set head radius

**Expected behavior**
Objects being able to fill the bed to the edge.

**Screenshots**
40mm radius
[image]

39mm radius
[image]

**Desktop (please complete the following information):**
Slic3r 1.41.1 PE under Linux.

lordofhyphens, 2018-11-23:
Please reproduce with main Slic3r 1.3.0 or 1.3.1-dev.

dl.slic3r.org/dev

Mirarkitty, 2018-11-25:
Which I can't. I guess the PE deviation is strong. I'll close this here.

closed 2018-11-25 by Mirarkitty

lordofhyphens, 2018-11-25:
That is why there is a note in the template that everything needs to be reproducible with main Slic3r. We don't have the resources to debug issues that only manifest in a fork.