slic3r/Slic3r #4614: Crash of Slic3r 1.3.0 in Kubuntu 18.04

jp0166, 2018-11-26:
1) At moment of run Slic3r-1.3.0-x86_64.AppImage: (Only warning. The program still run)
Warning message in Konsole:
(slic3r.pl:3118): Gtk-WARNING **: 20:55:15.973: Unable to locate theme engine in module_path: "adwaita",
(slic3r.pl:3118): Gtk-WARNING **: 20:55:15.984: Unable to locate theme engine in module_path: "adwaita",
Argument "" isn't numeric in numeric eq (==) at /tmp/.mount_Slic3rqkUT9k/usr/lib/local-lib/lib/perl5/Slic3r/GUI/MainFrame.pm line 103.

2) In any .stl model the program quit at the moment of click in Preview tab. The box of "Generate support material" was selected  
Error message in Konsole:
/tmp/.mount_Slic3rqkUT9k/AppRun: line 11:  3118 Aborted                 (core dumped) LD_LIBRARY_PATH="$DIR/usr/lib" "${DIR}/usr/bin/perl-local" -I"${DIR}/usr/lib/local-lib/lib/perl5" "${DIR}/usr/bin/slic3r.pl" --gui "$@"

System description: 
Kubuntu 18.04, (Ubuntu 18.04 with KDE plasma version 5.12.6.) Kernel 4.15.0-39-generic,  64 bit.

lordofhyphens, 2018-11-26:
You need to provide a model + config that Slic3r is crashing on; use Ctrl-E to produce the ini file as I can't reproduce this on my system. 

Also, try a development build: https://dl.slic3r.org/dev/linux

jp0166, 2018-11-26:
Version: Slic3r-1.3.1-dev-26b2a1e-x86_64.AppImage
[Files.zip](https://github.com/slic3r/Slic3r/files/2615586/Files.zip)
Warning message in Konsole:
(slic3r.pl:2700): Gtk-WARNING **: 07:53:40.866: Unable to locate theme engine in module_path: "adwaita",
(slic3r.pl:2700): Gtk-WARNING **: 07:53:40.878: Unable to locate theme engine in module_path: "adwaita",

With "Generate support material"  (box in rigth part of GUI), the program quit.
Error message in Konsole:
/tmp/.mount_Slic3rnHBDN9/AppRun: line 11:  2700 Aborted                 (core dumped) LD_LIBRARY_PATH="$DIR/usr/lib" "${DIR}/usr/bin/perl-local" -I"${DIR}/usr/lib/local-lib/lib/perl5" "${DIR}/usr/bin/slic3r.pl" --gui "$@"

lordofhyphens, 2018-11-26:
FYI, the warnings can be ignored.

lordofhyphens, 2018-11-26:
Backtrace:
````
#0  0x00007f8f69fb1f3b in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007f8f69fb32f1 in __GI_abort () at abort.c:79
#2  0x00007f8f6874fdee in _Unwind_SetGR () at /home/lenox/Slic3r-builds/Slic3r-26b2a1e/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#3  0x00007f8f686f9d21 in __gxx_personality_v0 () at /home/lenox/Slic3r-builds/Slic3r-26b2a1e/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#4  0x00007f8f6a36bd73 in  () at /lib/x86_64-linux-gnu/libgcc_s.so.1
#5  0x00007f8f6a36c2d1 in _Unwind_RaiseException () at /lib/x86_64-linux-gnu/libgcc_s.so.1
#6  0x00007f8f65b8aaf7 in __cxa_throw () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007f8f65b86851 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007f8f686b751e in Slic3r::Print::max_allowed_layer_height() const ()
    at /home/lenox/Slic3r-builds/Slic3r-26b2a1e/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#9  0x00007f8f683282f3 in XS_Slic3r__Print_max_allowed_layer_height(interpreter*, cv*) ()
    at /home/lenox/Slic3r-builds/Slic3r-26b2a1e/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#10 0x00000000004c0050 in Perl_pp_entersub ()
#11 0x00000000004b8736 in Perl_runops_standard ()
#12 0x000000000043af62 in Perl_call_sv ()
#13 0x00007f8f6705efed in wxPliEventCallback::Handler(wxEvent&) () at /home/lenox/Slic3r-builds/Slic3r-26b2a1e/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so
#14 0x00007f8f65d3babe in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const () at ./bin/libwx_baseu-3.0.so.0
#15 0x00007f8f65ecd327 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at ./bin/libwx_baseu-3.0.so.0
#16 0x00007f8f65ecd706 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at ./bin/libwx_baseu-3.0.so.0
#17 0x00007f8f65ecd78e in wxEvtHandler::TryHereOnly(wxEvent&) () at ./bin/libwx_baseu-3.0.so.0
#18 0x00007f8f65ecd853 in wxEvtHandler::ProcessEventLocally(wxEvent&) () at ./bin/libwx_baseu-3.0.so.0
#19 0x00007f8f65ecd8b5 in wxEvtHandler::ProcessEvent(wxEvent&) () at ./bin/libwx_baseu-3.0.so.0
#20 0x00007f8f65ecea20 in wxEvtHandler::ProcessPendingEvents() () at ./bin/libwx_baseu-3.0.so.0
#21 0x00007f8f65d3f7a7 in wxAppConsoleBase::ProcessPendingEvents() () at ./bin/libwx_baseu-3.0.so.0
#22 0x00007f8f663fbef2 in wxApp::DoIdle() () at ./bin/libwx_gtk2u_core-3.0.so.0
#23 0x00007f8f663fc033 in wxapp_idle_callback () at ./bin/libwx_gtk2u_core-3.0.so.0
#24 0x00007f8f6548dae8 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#25 0x00007f8f6548ded8 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#26 0x00007f8f6548e1d2 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#27 0x00007f8f657e18e7 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#28 0x00007f8f6641d065 in wxGUIEventLoop::DoRun() () at ./bin/libwx_gtk2u_core-3.0.so.0
#29 0x00007f8f65d7fc90 in wxEventLoopBase::Run() () at ./bin/libwx_baseu-3.0.so.0
#30 0x00007f8f65d3dc46 in wxAppConsoleBase::MainLoop() () at ./bin/libwx_baseu-3.0.so.0
#31 0x00007f8f66f93d79 in wxPliApp::MainLoop() () at /home/lenox/Slic3r-builds/Slic3r-26b2a1e/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so
#32 0x00007f8f66f47141 in XS_Wx__App_MainLoop(interpreter*, cv*) () at /home/lenox/Slic3r-builds/Slic3r-26b2a1e/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so
#33 0x00000000004c0050 in Perl_pp_entersub ()
#34 0x00000000004b8736 in Perl_runops_standard ()
#35 0x0000000000442889 in perl_run ()
#36 0x000000000041cb9b in main ()
````

lordofhyphens, 2019-01-06:
@jp0166 looks like it's working as of https://dl.slic3r.org/dev/linux/Slic3r-1.3.1-dev-b3cac9d-x86_64.AppImage on my Xubuntu 18.04 VM. Closing.

closed 2019-01-06 by lordofhyphens

jp0166, 2019-01-07:
This file (Slic3r-1.3.1-dev-b3cac9d-X86_64.AppImage)  is working in my computer. Thanks a lot.