slic3r/Slic3r #4615: supports overlap object when 2 STL files loaded

csimmerling, 2018-11-26:
**Describe the bug**
When I split an STL file into 2 objects and then generate supports, the supports for the first object can overlap and go through the second object. This results in inability to remove supports

**To Reproduce**
Load STL file with 2 objects that don't overlap but do overlap on the Z axis
split object into 2
slice with automatic supports
the supports for the higher object go through the lower object

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.

**Expected behavior**
when supports are generated, both objects should be considered and not sequential - meaning that the first object supports can be over or under the second, but not inside it.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 1.41.2 (PE)
     - _Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._
     - _For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     - _Use `About->About Slic3r` for release versions._

**Additional context**
Add any other context about the problem here.

lordofhyphens, 2018-11-27:
@csimmerling

```Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. Only use normal Slic3r version IDs.
````

Will reopen when it is confirmed that this is a problem on 1.3.0 (dl.slic3r.org) or 1.3.1-dev (dl.slic3r.com/dev)

csimmerling, 2018-11-29:
confirmed that the problem exists in 1.3.1-dev. screenshot below.

[image]

csimmerling, 2018-11-30:
confirmed to be a problem on 1.3.1-dev, please reopen (uploaded new
information to thread)