slic3r/Slic3r #4619: Starting Slic3r fails on Win 10 x64 will latest versions

vag2t, 2018-11-27:
Hello to all,

I consistently fail to start the program on my home PC (a Win 10 Pro x64).
I tried watching the process in task manager and while I see the slicer.exe, after a couple of seconds it disappears. 
In general everything is up to date and there is no antivirus running except the Defender (deactivating doesn't help). 
Also tried opening from different file locations, with win7-8 compatibility and admin rights - to no avail.

This happens will all versions from 1.3 back to 1.2.5 (x64).
Trying to open the x86 version works with 1.2.9, but not with 1.3 - haven't tried another.
Both x86 & x64 latest (27/11) dev. versions  won't run either.
Finally, both the slicer-console executables will not start also.

At work (Win 7 x64) -similar localization settings- the program starts just fine.
Would there be anything to try and resolve the issue?
Thanks in advance! :)

supermerill, 2018-11-28:
You can try to delete/rename your local settings.

vag2t, 2018-11-29:
Thanks for the assistance.
Running through the command prompt doesn't seem to change something, nor presents an error message.

By deleting/changing the local settings you mean the temporary settings/files of the OS ?
Deleting those and restarting didn't help either.