slic3r/Slic3r #4620: Top and bottom solid layers vanish when changed

djdelorie, 2018-11-28:
**Describe the bug**
Whenever I change the number of top or bottom solid layers, either through retyping or up/down arrows, all existing top and bottom solid layers vanish.  Independent of tab/dialog setting.  Independent of background processing.

**To Reproduce**
Load any model.  A test cube will do.  Use a config that has top and bottom layers. 
Change the number of top layers.
See the top/bottom layers vanish

[missing-tb-layers.zip](https://github.com/slic3r/Slic3r/files/2622904/missing-tb-layers.zip)

**Expected behavior**
Number of top and bottom layers should match config

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Fedora 26 x86-64
 - Version: git 33b3faacfd3b50cc643d6eecd6f48aec8ddcc7bc

lordofhyphens, 2018-11-28:
From my bisect, it's reproducible without using background slicing as well.
````
~/slic3r$ git bisect bad
f2e2c9aa71208273427eb124281771fe7e00c6d1 is the first bad commit
commit f2e2c9aa71208273427eb124281771fe7e00c6d1
Author: Alessandro Ranellucci <aar@cpan.org>
Date:   Mon Nov 26 21:16:37 2018 +0100

    Ported PrintObject::make_perimeters() to C++

:040000 040000 2479db9122a97ac5307b4bfd92a8781f8f2b7e07 88ebed84c2c495b29579fdd7d38e8f851cf11d8d M	lib
:040000 040000 b5145b7b0018b6816854ccdc2e53ec2630b13f6a 8bd6e5216a2a97342e83f4774e8322ed8a105cb6 M	xs
````

lordofhyphens, 2018-11-28:
Debug statements sprinkled in to the print setup.

````
Calling Print::Process
Pos infill state done? DEBUG: 0
PrintObject DEBUG: Invalidating perimeters.
PrintObject DEBUG: Making perimeters.
Pos infill state done? DEBUG: 1
Pos infill state done? DEBUG: 1
Calling Print::Process
Pos infill state done? DEBUG: 0
PrintObject DEBUG: Invalidating perimeters.
PrintObject DEBUG: Making perimeters.
Pos infill state done? DEBUG: 1
Pos infill state done? DEBUG: 1
````

VanessaE, 2018-12-09:
Happens for me, also..

djdelorie, 2019-01-06:
also happens with today's git if I just change the number of perimeters

lordofhyphens, 2019-01-06:
As far as I can tell, during `make_perimeters()` the slice is supposed to be invalidated if there were merged surfaces (this is a workaround for another method not being idempotent). This is fine.

However, `slice()` is forcing `typed_slices` to `false` and the surface detection is never re-run because that step is never invalidated here and there's a block stopping it from being run later. 

What I'm going to do is force the invalidation of `posDetectSurfaces` if we're in that state with `typed_slices = true` when make_perimeters() is called. This should ensure that it's invalidated along with the slice in these situations.

PR #4679 (2019-01-07): Patch for #4620 - ensure posDetectSurfaces is invalidated on slice change
Invalidate posDetectSurfaces during make_perimeters() when posSlice would also get invalidated.
Can't use invalidate_step() as that invalidates the perimeters as well.

Fixes #4620
closed 2019-01-07 by lordofhyphens