slic3r/Slic3r #4623: Lumps on a thread

FlailAway, 2018-11-29:
Hi, This is a cap for a propane tank and it has an Acme-thread.

Latest version of three Slicers, (.appimage) running under Ubuntu 16.04.

From Left to Right, the pic shows sliced with Cura, Slic3r-PE and Slic3r.

Slic3r and Slic3r-PE both have lumps on the threads. All three have the same settings for Layers, speeds, Retraction etc.

I **think** it has something to do with Slic3r's Retraction (5mm at 40mm/s) as when printing, it pauses momentarily about where each lump is.

I really prefer Slic3r to Cura but obviously this is not an option. Is this something I can tune out or not? If I can, how?

Thanks
(I originally posted this to RepRap forums)
[image]

supermerill, 2018-11-29:
Can you show the difference in the g-code, at these points?

FlailAway, 2018-11-29:
How do I find that in the gcode file it is 2MB long. Is there some way to trace it through in Cura/Slic3r?

supermerill, 2018-11-29:
You can print slic3r with comments, to know where it retract, where the layer change.

Choose a layer.  (or 3), slice, print.

Note the moment where it's problematic (after perimeter, external, internal? in the gapfill?, in the infill?), then retreive it inside the commented gcode.

FlailAway, 2018-11-29:
I have cut just-over one full turn of the Acme thread and will slice and print that, but Exporting to .3mf crashed the entire thing.

supermerill, 2018-11-29:
well, I hope the g-code will be enough. Can you export the cut stl (only)?

FlailAway, 2018-11-29:
OK, it's not retraction, my mistake, an assumption as it seemed to make lumps at the visual point it paused for retracting. I turned Retraction off and got a lot of stringing but still got the lumps.

Why all the speed changes? I have it set at 120mm/sec.
The lumps may be caused by oozing while stationary for quarter-second..

First layer
G1 X26.174 Y-4.142 E27.41363 ; skirt
G1 X26.301 Y-3.231 E27.47512 ; skirt
G1 E22.47512 F2400.00000 ; retract extruder 0
G92 E0 ; reset extrusion distance
G1 X3.926 Y-14.436 F12000.000 ; move to first perimeter point
G1 E5.00000 F2400.00000 ; unretract extruder 0
G1 F1800
G1 X4.257 Y-14.345 E5.02295 ; perimeter
G1 X5.173 Y-14.040 E5.08745 ; perimeter

Part way through the first layer it crosses to the other side for some reason
G1 X3.868 Y-14.452 E11.88777 ; perimeter
G1 X3.783 Y-13.920 F12000.000 ; move to first external perimeter point
G1 F1800
G1 X4.101 Y-13.832 E11.90980 ; external perimeter
G1 X4.991 Y-13.537 E11.97241 ; external perimeter

I think this is where it is happening. Why move inwards?
G1 X3.725 Y-13.936 E18.55063 ; external perimeter
G1 X4.016 Y-14.153 F12000.000 ; move inwards before travel
G1 E13.55063 F2400.00000 ; retract extruder 0
G92 E0 ; reset extrusion distance
G1 X-17.506 Y-5.405 F12000.000 ; move to first perimeter point
G1 E5.00000 F2400.00000 ; unretract extruder 0
G1 F1800
G1 X-17.772 Y-5.515 E5.01923 ; perimeter

FlailAway, 2018-11-30:
> Do you have wipe turned on?

Nope. Thanks for looking at this

FlailAway, 2018-12-01:
> Also, please provide a copy of the ctrl-e config export and a copy of the model (zip and upload here).

Hi I edited my reply to your question about "wipe," as I had forgotten to attach the .zip. Now not sure if you saw the early Reply and no file and have moved on. Probably should have done a new Reply for adding the .zip. 

I am back to using Cura and that sucks as under Linux, the Cura File handling is abysmally cumbersome. I really would like to get back to Slic3r.

FlailAway, 2018-12-05:
Bump. Anyone? Anything?

lordofhyphens, 2018-12-05:
@FlailAway we're all volunteers here. I haven't had time to look at this issue any closer.

FlailAway, 2018-12-06:
I understand that and appreciate the help, but the "thanks" was very non-committal, had you written, "thanks, give me some time to get to it," I would check back daily, but wait weeks before the "bump."

I am just a little antsy having to use Cura. It's all good though, so take your time I will harass no more. :)