slic3r/Slic3r #4624: No gap fill on first layer

geofox784, 2018-11-29:
**Describe the bug**
There does not appear to be any gap fill on the first layer. If a line does not completely fit in the infill area it will be skipped resulting in a gap. This prevents the print from being watertight. 

**To Reproduce**
Model with slight curves at base perimeter that has those curves parallel to the fill angle. Can be temporarily fixed for a particular model by adjusting the fill angle, but I slice many models at once via a batch command so this is not feasible. Increasing Infill/perimeters overlap does not fix the problem either. If an infill line crosses a perimeter line it will still be cut even with overlap increased. 

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 1.3.1-dev 
    Latest developmental build. Reproducible on stable version as well.

supermerill, 2018-11-29:
There are no gapfill in a rectilinear infill, as far as i know (there are multiples version of rectilinear). No gapfill in any infill.

geofox784, 2018-11-30:
If it is not a bug, then maybe this could be added a a new feature? It definitely seems logical to have gap fill on outer infill layers, especially the bottom one.