slic3r/Slic3r #4626: Gcode command to print commanded pressure

mngr0, 2018-11-29:
Hi, 
I was thinking about a different way to manage the 3d printer, in particular regarding the extruder axis and pressure.
Slic3r decides the pressure needed in every moment and from this calculates the extruder movement.

Is there any way to print directly the pressure requested? 
It should output a command like M180 S4 that is supposed to be a new M command.
It should output the necessary pressure and let the printer manage the extruder motor and retraction.
S4 is the required pressure, in whatever units it needs to be.

I just want to calculate the extruder movement directly on the printer, but I don't want to calculate pressure from extruder movement, so I need to have the pressure required written in the GCODE, somehow...

This means making changes in slic3r, is a thing like this possible?

regards, 
mngr0

supermerill, 2018-11-29:
Something like that? https://github.com/slic3r/Slic3r/pull/3734

mngr0, 2018-11-29:
yes! Wow, that was faster than expected!
Ok, now I will go study in detail how to do it on the printer side!
Thanks!

closed 2018-11-29 by mngr0

mngr0, 2018-11-30:
Where is declared the autospeed module?