slic3r/Slic3r #4627: Crashes with libGL error

simonCor, 2018-11-30:
**Describe the bug**
Start of the AppImage crashes with the following error message:
```
libGL error: unable to load driver: radeonsi_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: radeonsi
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 2009 error_code 2 request_code 152 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```
It seems to be connected to the open AMD drivers. The problem started to happen when I interchanged my NVIDIA graphics card to a AMD (RX570) one.

**To Reproduce**
Just start the AppImage: ./Slic3r-1.3.0-x86_64.AppImage

**3D Model and Slic3r Configuration Export**
Not related to a model.

**Expected behavior**
The program should start.

**Desktop (please complete the following information):**
 - OS: openSUSE Leap 15.0
 - Version 1.3.0

**Additional context**
I found a workaround on the internet which works for me: https://superuser.com/questions/1377555/slic3r-appimage-fails-to-run-with-libgl-error-unable-to-load-driver-radeonsi/1377556

When I start it with 
```
LD_PRELOAD='/usr/lib64/libstdc++.so.6' ./Slic3r-1.3.0-x86_64.AppImage
```
Slic3r works.

lordofhyphens, 2018-12-01:
Ugh, that means it'd break old distros if I removed it.

lordofhyphens, 2019-01-06:
@simonCor Give one of the recent builds from https://dl.slic3r.org/dev/linux a try. 
libstdc++.so.6 should only get pulled in now if it's actually necessary.

closed 2019-01-06 by lordofhyphens

simonCor, 2019-01-06:
Tested (Build commit: b3cac9d) works.